Clon1998/mobileraker #7: Moonraker power control

viljamip, 2021-08-18:
Possibilty to turn devices configured under Moonraker’s [power] section on or off. Currently, the UI does not load of Klipper is shutdown due to unpowered MCU. The power controls should be accessible even if Klipper is not running.

katosabi, 2021-09-20:
I'll second this. I use 2 relays to control both MCU and bed power via moonraker, and I have to turn these on via web before I can begin working with the printer in mobileraker.

Config info:
https://moonraker.readthedocs.io/en/latest/configuration/#power

nightime7, 2022-10-01:
I would also add here that even if there is no MCU available, I'd still like the camera feed to work in the app since many times, when I finish a print, the printer turns off the MCU but leaves the RPi on so I can see what the results were and obviously turn it back on as needed.

Clon1998, 2022-10-04:
Added in 2.0.0 via 89797faed54ac599b72b65a4b018971ce994d4fc

closed 2022-10-04 by Clon1998