Clon1998/mobileraker #17: Timer alarm next to preheat button

joshtownsend66, 2021-09-12:
Everytime i preheat a voron i pull out and set my timer to 5 minutes so ill know when to start. I want to be there to check the skirt at the start of a print. I would pay for this feature

Clon1998, 2021-09-12:
So you want to get a notification once the printer is preheated through one of the presets?

joshtownsend66, 2021-09-12:
That would work. I was thinking maybe linking it to the OS timer/clock or
something but anything like that. I forget to set my timer sometimes,
besides all the steps to get there and do it.

Clon1998, 2022-12-18:
I will close the issue since this will be possible via the new custom `M117` user notifications supported through mobileraker_companion.
For more info and how to test the beta visit: https://github.com/Clon1998/mobileraker_companion/blob/dev/Custom_Notifications.md

closed 2022-12-18 by Clon1998