Clon1998/mobileraker #20: Publish in F-Droid?

dstrelnikov, 2021-11-06:
Hi! Thank you very much for this cool app!

Could you please consider publishing it in F-Droid?

Clon1998, 2022-04-10:
Sadly I rely on Google Firebase Cloud messaging for push notifications.

closed 2022-04-10 by Clon1998

gebbebebbebebr, 2023-06-18:
Could you take a another look at perhaps getting an F-Droid release going? From what I've seen release [2.1.0](https://github.com/Clon1998/mobileraker/releases/tag/android2.1.0) changed the way notifications are handled so it shouldn't be a problem anymore to have the main app on F-Droid right?

Clon1998, 2023-06-18:
> Could you take a another look at perhaps getting an F-Droid release going? From what I've seen release [2.1.0](https://github.com/Clon1998/mobileraker/releases/tag/android2.1.0) changed the way notifications are handled so it shouldn't be a problem anymore to have the main app on F-Droid right?

Hey, 
sadly, due to the[ use of Firebase](https://github.com/Clon1998/mobileraker/blob/a8b221b0fef9490978106c5c37e7843ed13f0068/pubspec.yaml#L34-L39) and the google play services, I cannot release it via F-Droid. (See [requirements](https://f-droid.org/en/docs/Inclusion_Policy/) of F-Droid)

tinyfluffs, 2023-08-23:
It was highlighted to me that UnifiedPush would be a good means of getting it onto F-Droid. I'd suggest taking a look into that.

https://unifiedpush.org/

Clon1998, 2023-08-23:
As stated before, being the only dev of the project I dont have to capacity to work on an F-Droid compliant version. Quite early I decided to go with apple+google services / firebase because it simplifies the development.
I've been approached now by multiple people including some university groups that asked me to publish an F Droid version. However, I simply don't have the resources to adapt the app's code for F-Droid. In addition, due to the license change of the project a few months ago an F Droid release is not possible.

tinyfluffs, 2023-08-23:
That's entirely fair. I'd look into it myself, but I've been super rusty with Android development for quite a number of years at this point. However, even just cutting out push notifications altogether couldn't be too much work. Depends on the state of the codebase, of course.

Clon1998, 2023-08-23:
> That's entirely fair. I'd look into it myself, but I've been super rusty with Android development for quite a number of years at this point. However, even just cutting out push notifications altogether couldn't be too much work. Depends on the state of the codebase, of course. Nonetheless, I've appreciated having Mobileraker around.

At this point it's not just push notifications.
To make my life easier the app is now also using Crashalytics for error reporting and analytics. As well as some 3p to process IAP/Subscriptions. The codebase itself is also not pure android, rather it is using flutter.

tinyfluffs, 2023-08-23:
Oh, that is extremely disappointing, and somewhat scary in fact that it's not even a fully open source model anymore. It super frustrates me that folks like Thomas Sanladerer have been spreading this mass panic to ditch open source for closed source and/or patent models. Please don't follow in the footsteps of Slice Engineering and E3D.

Clon1998, 2023-08-24:
> Oh, that is extremely disappointing, and somewhat scary in fact that it's not even a fully open source model anymore. It super frustrates me that folks like Thomas Sanladerer have been spreading this mass panic to ditch open source for closed source and/or patent models. Please don't follow in the footsteps of Slice Engineering and E3D. If you wish to prevent commercial adoption without publication of source, there's AGPL.

The project is still open source. Only commercial use is limited.

Clon1998, 2023-08-24:
And also, I have spent probably thousands of hours at this point working on the app. So protecting my IP is more than legit.

Jookia, 2023-08-24:
Could you please no longer mark your application as 'free and open source' then? It's confusing as it means https://en.wikipedia.org/wiki/Free_and_open-source_software which this project doesn't fit. You need to be clear with people what their rights are for the software.

dstrelnikov, 2023-08-24:
As the author of this feature request I feel it's time for me to stand in for @Clon1998.

It is very clear from the log that Patrick is carrying the project on his shoulders alone. One may suggest the stack of technologies to use for the project but Patrick has the right to make decisions based on his own interests. People don't realize that how much it takes to develop and maintain such a project.

The app is high quality and fully functional free of charge. The source code is fully available and therefore it is an *open source* project. If the license is not compatible with the conventional definition of FOSS it does not immediately imply that there is no value in the project.

@Clon1998, perhaps you should change this phrase in the README
>Mobileraker is free and open source

to something that is less likely to be misinterpreted as FOSS, like
> Mobileraker is open source and free for non-commercial application

I would also suggest to chose one of the standard licenses that would reflect your own ideas about this project the best. The motivation behind it is that such licenses have been reviewed by large numbers of people and perhaps even tested in the court multiple times, so it is usually a more solid and safe choice to use one of the standard options. But ultimately it is up to you.

Regarding the third party services. If the core of the app is still functional on devices without google play services (i.e. just some features don't work), you could just build-in a little updater to download the latest build from Github, like how it's done in [Cromite](https://github.com/uazo/cromite). But again, it's your choice.

You made a wonderful app and I would like to thank you again! Please don't let other people discourage you just because of their expectations.

tinyfluffs, 2023-08-25:
Marking our prior discussions here as off-topic is extremely childish and uncalled for. There was good value in what was being said, and it reads to me that you cannot take constructive feedback. Quite frankly, the attitude stinks. I would appreciate if you took some time for some self reflection.

I've uninstalled Mobileraker. I'm disappointed. There was immense value to having a fully open platform supported by yourself, but instead you are more concerned about protecting an "IP". Which, bear in mind, was in the public sphere until June. So I have forked the project from before the license change, and I will pick it up in mid September for F-droid under a different name. I will continue to uphold the original MIT license, and I will continue to credit your original work.

Despite our differences, I wish you good luck with the future.

dstrelnikov, 2023-08-25:
Hey guys, let's find a common ground instead of creating more fragmentation in the open source community.

@tinyfluffs, I agree that marking the comments as off-topic was too much. But before saying that you are disappointed please have a look into the git log. There is almost a thousand commits from @Clon1998 and maybe 5-10 feature/bug related commits from other people besides the translations. I think it's also a bit too harsh to say that you are disappointed while someone put so much effort into creating something nice and useful. 

I see the point with the license concerns and how it makes the project community unfriendly. But if you fork the project prior to the license change and publish it under a new name, you will achieve the following:
- @Clon1998 will see happening exactly what he was trying to avoid, i.e. the conflict will further escalate 
- more fragmentation will be created, different people will commit to different projects (again, so far @Clon1998 is the only essential contributor) slowing down the progress

I suggest to have a civil discussion and suggest a community friendly license that will protects the interests of @Clon1998 as well as protecting the interests of the community.

@tinyfluffs, in my opinion, if you are ready to commit to developing the fork by yourself, you could also suggest some changes in the code base in a form of a merge request. I've seen people requesting the author to implement certain things the way they like, but no one has provided a merge request or a bounty. @Clon1998 clearly said that the reason for his decisions was to make his life easy because he is at the moment carrying the project alone. I suggest consolidating your efforts in order to make not more software but better software.

Clon1998, 2023-08-25:
> Marking our prior discussions here as off-topic is extremely childish and uncalled for. There was good value in what was being said, and it reads to me that you cannot take constructive feedback. [...]

I don't want to go too deep into that another time. So this will be my only response to that.

First of all best of luck with F-Droid. As you mentioned it was licensed under MIT and therefore you are free to do that. I am sure it will bring value to the Community. Let me know if you need help setting it up.

However, claiming that hiding our comments about the project's license and my move to a different license is childish and that I am unable to take constructive feedback is simply a false claim. This issue was about releasing the app in F-Droid not about the license of the project. Therefore, our discussion was off-topic. You yourself could have chosen the path that @Jookia took in #231 and simply created a new issue to discuss all of this. This is also true for me. Therefore, hiding those comments was more than valid IMO. Also, they are just hidden and not deleted.

Clon1998, 2023-08-25:
> Hey guys, let's find a common ground instead of creating more fragmentation in the open source community.
> 
> @tinyfluffs, I agree that marking the comments as off-topic was too much. But before saying that you are disappointed please have a look into the git log. There is almost a thousand commits from @Clon1998 and maybe 5-10 feature/bug related commits from other people besides the translations. I think it's also a bit too harsh to say that you are disappointed while someone put so much effort into creating something nice and useful.
> 
> I see the point with the license concerns and how it makes the project community unfriendly. But if you fork the project prior to the license change and publish it under a new name, you will achieve the following:
> 
> * @Clon1998 will see happening exactly what he was trying to avoid, i.e. the conflict will further escalate
> * more fragmentation will be created, different people will commit to different projects (again, so far @Clon1998 is the only essential contributor) slowing down the progress
> 
> I suggest to have a civil discussion and suggest a community friendly license that will protects the interests of @Clon1998 as well as protecting the interests of the community.
> 
> @tinyfluffs, in my opinion, if you are ready to commit to developing the fork by yourself, you could also suggest some changes in the code base in a form of a merge request. I've seen people requesting the author to implement certain things the way they like, but no one has provided a merge request or a bounty. @Clon1998 clearly said that the reason for his decisions was to make his life easy because he is at the moment carrying the project alone. I suggest consolidating your efforts in order to make not more software but better software. It is always possible to make a fork as the last resort.

To highlight @dstrelnikov point, here are the git stats of mobileraker, exported via [git-fame](https://github.com/casperdcl/git-fame):

```
Total commits: 970
Total ctimes: 10756
Total files: 543
Total loc: 653400
| Author            |    loc |   coms |   fils |  distribution (loc, coms, files)  |
|:------------------|-------:|-------:|-------:|:----------------|
| Patrick Schmidt   | 646677 |    931 |    516 | 99.0/96.0/95.0  |
| Livex97           |   1146 |      1 |      1 | 0.2/ 0.1/ 0.2   |
| vaxxi             |   1144 |      1 |      1 | 0.2/ 0.1/ 0.2   |
| Jothoreptile      |   1121 |     10 |      1 | 0.2/ 1.0/ 0.2   |
| teuchezh          |   1121 |      3 |      1 | 0.2/ 0.3/ 0.2   |
| Austin S          |    554 |      3 |      9 | 0.1/ 0.3/ 1.7   |
| Males Tomlinson   |    514 |      1 |      2 | 0.1/ 0.1/ 0.4   |
| AntoszHUN         |    512 |      6 |      1 | 0.1/ 0.6/ 0.2   |
| JMSPI             |    512 |      1 |      2 | 0.1/ 0.1/ 0.4   |
| Jochen Ullrich    |     74 |      5 |      7 | 0.0/ 0.5/ 1.3   |
| ptsa              |     24 |      2 |      1 | 0.0/ 0.2/ 0.2   |
| nosoymilk         |      1 |      3 |      1 | 0.0/ 0.3/ 0.2   |
| Björn Marschollek |      0 |      1 |      0 | 0.0/ 0.1/ 0.0   |
| Manuel Raimo      |      0 |      1 |      0 | 0.0/ 0.1/ 0.0   |
| emo64             |      0 |      1 |      0 | 0.0/ 0.1/ 0.0   |
```

Jookia, 2023-08-25:
> However, claiming that hiding our comments about the project's license and my move to a different license is childish and that I am unable to take constructive feedback is simply a false claim. This [...]

... Your discussion about push notifications and analytics that would prevent it from being uploaded to F-Droid was off topic?

tinyfluffs, 2023-08-25:
> @tinyfluffs, I agree that marking the comments as off-topic was too much. But before saying that you are disappointed please have a look into the git log. There is almost a thousand commits from [...]

My disappointment was not with the project itself, but the direction in which it is heading with the new license terms. I have heard nothing but great things regarding Mobileraker, and picked it up this year after so many people kept telling me about it. I love the work that has been done, and I really do value it being here.

> I see the point with the license concerns and how it makes the project community unfriendly. But if you fork the project prior to the license change and publish it under a new name, you will achieve [...]

The issues Clon discuss would then be problems that I would have to face. Quite frankly, I don't care all that much if Creality came around and slapped a badge on the work and called it a day. But I personally see the answer to this is to maintain an open source license which is either the MIT license itself, or a compatible license such as AGPL. Companies violating the terms of a license agreement does not stop the problem if we change license anyway - they will just do it in secret (see KlipperScreen on the Creality K1). Yes, it would cause more fragmentation, and I would honestly prefer that we do it under one name. But in order for me to feel comfortable contributing upstream, I would like the license to remain MIT.

> I suggest to have a civil discussion and suggest a community friendly license that will protects the interests of @Clon1998 as well as protecting the interests of the community.

That is exactly what I have been advocating for this entire time.

> @tinyfluffs, in my opinion, if you are ready to commit to developing the fork by yourself, you could also suggest some changes in the code base in a form of a merge request. I've seen people [...]

As I said, I would be willing to invest the time if the license was more open.

dstrelnikov, 2023-08-25:
@Clon1998, it looks like we are on a good way. Could you please give more details about your concerns regarding the license, so we could look into it together to find the best solution?

> But I personally see the answer to this is to maintain an open source license which is either the MIT license itself, or a compatible license such as AGPL. Companies violating the terms of a license [...]

I generally agree that a proprietary license will not protect you from indecent companies but rather turn out the community (people who would like to contribute to your project also want to do it on clear terms, i.e. their time investments being protected).

Jookia, 2023-08-26:
I don't want to spoil the party but it's not possible to change the license again at this point: Third party contributions have been accepted under it and there is no copyright assignment. The current license is staying forever.