Clon1998/mobileraker #21: Cannot build - missing includes

hanzov69, 2021-11-17:
**Describe the bug**
Attempting to build pre-1.4 for iOS on local machine. Appears a few includes are missing.

**To Reproduce**
Steps to reproduce the behavior:
1. Setup build environment (flutter, cocoapods, Xcode, etc)
2. Launch simulator (`open -a Simulator`)
3. attempt build (`flutter run`)
4. See error

**Additional context**
I've attached the build output
[build.log](https://github.com/Clon1998/mobileraker/files/7555843/build.log)

Clon1998, 2021-11-17:
You didnt run the build runner as stated in the **Getting Started** section of the readme.

> Getting Started
After importing this project into your IDE be sure to run `flutter packages pub run build_runner build` in order to generate required files!

closed 2021-11-17 by Clon1998

hanzov69, 2021-11-17:
🤦‍♂️ 
d'oh!

Thanks!