Clon1998/mobileraker #22: `Unhandled method registerPeriodicTask`

hanzov69, 2021-11-17:
**Describe the bug**
I hope I'm not missing something obvious here, just trying to get the iOS build up and running locally. 

While it builds without exception, when running in the iOS simulator, it will prompt to allow notifications, then show a white screen. Running it in VSCode with the flutter extension, it breaks on main.dart:32, `PlatformException (PlatformException(unhandledMethod("registerPeriodicTask") error, Unhandled method registerPeriodicTask, null, null))
`

A little searching reveals that `registerPeriodicTask` isn't supported for iOS. 
https://github.com/fluttercommunity/flutter_workmanager#customisation-android-only

Admittedly, I'm out of my depth with this, so if this is obvious/I'm doing something stupid, apologies in advance.
[image]

Clon1998, 2021-11-19:
I changed the code to ensure that it will only be used for Android in fe562464f6482bb918de7122d0d524f20e9062ab

hanzov69, 2021-11-20:
Pulled and built- fixed!

Anticipated, I assume, but notifications only seem to work when app is open/active, background notifications don't seem to appear.

closed 2021-11-20 by hanzov69