Clon1998/mobileraker #24: Layer is not showing the current layer

Syco54645, 2021-11-29:
**Describe the bug**
Layer is not showing the current layer and seems it stays at Max/Max

**To Reproduce**
Steps to reproduce the behavior:
1. Start a print
2. Open mobileraker
3. Observe Layer area

**Expected behavior**
This should update with the current layer/max layer

**Smartphone (please complete the following information):**
 - Device: Pixel 5a
 - OS: Android 12

**Additional context**
Add any other context about the problem here.

Clon1998, 2021-11-29:
Can you also provide the version of Mobileraker youre using?
For Android, it can be found in Android's App-Info (Where you can set permissions, clear storage...).

Syco54645, 2021-11-29:
> Can you also provide the version of Mobileraker youre using?
> For Android, it can be found in Android's App-Info (Where you can set permissions, clear storage...).
> 

Done. 

While I have your attention I just want to say thanks. This application is awesome and very much needed. Keep up the great work.

Clon1998, 2021-11-29:
I am glad you like it.
I am not sure at what time I fixed this issue but I remember fixing it. 
Version 1.3 definitely will fix this issue. If you're in the VOC you could check the mobileraker channel for the latest pre-version APK.

Syco54645, 2021-12-03:
I am not in the voc and am unfamiliar. What is it and how can I join?