Clon1998/mobileraker #26: Webcam does not load

simpat1zq, 2021-12-03:
I'm on android with the latest app version. I added 2 printers and then added their webcams, and it just doesn't load. I tried the exact same URL in the web browser and it brings up the stream.

Clon1998, 2021-12-21:
Sorry for answering that late, I somehow missed this.
Can you verify that the webcam streams are MJPEG ones?

simpat1zq, 2021-12-21:
Yes. I actually have 3 printers. It's working fine on one, not working on another, and for some reason the app doesn't connect to the third one at all any more. The one that works fine has a USB webcam, and the one that doesn't has the PI camera. All are using mjpgstreamer. 

And on a side note, the one that doesn't even connect is able to load the files just fine, it's the overview page that doesn't load.

Clon1998, 2021-12-21:
Someone had a similar issue regarding the connectin problem only in overview, and it was fixed by using the latest dev build.
Give me this week and hopefully, the latest build will be ready.

simpat1zq, 2021-12-21:
I'm on the pre-1.4 build. Not sure if that's the dev build you are referring to.

Clon1998, 2021-12-21:
> Someone had a similar issue regarding the connectin problem only in overview, and it was fixed by using the latest dev build. Give me this week and hopefully, the latest build will be ready.

I think so. Cant remember which version the other person used. Ill think about a way to debug this issue....

closed 2022-03-31 by Clon1998