Clon1998/mobileraker #29: Super laggy when webcam is added

Aasikki, 2021-12-21:
**Describe the bug**
The interface has very low fps when camera is added, so low that it becomes unusable.

**To Reproduce**
Steps to reproduce the behavior:
1. Add camera in settings
2. Restart mobileraker as the camera won't load otherwise
3. Lag

**Expected behavior**
-

**Screenshots**
Demonstration video: https://youtu.be/di9L0-6G0jo

**Smartphone (please complete the followininformation):**
 - Device: OnePlus 7 Pro
 - OS: Android 11 (Havoc OS v4.11)
 
**Mobileraker-Version
1.2.1 (4)

**Additional context**
My webcam is raspberry pi camera set to 1280x720 30fps

Clon1998, 2021-12-21:
I can not reproduce this with my device. My guess is, that this is related to #10. Since the lib I am using doesn't drop missed frames and instead tries to keep up all of the UI starts to lag. Could you try to reduce the fps to 10 and see if this solves the issue?

Aasikki, 2021-12-23:
I tried dropping the framerate to 10 and it got much better but still not perfectly smooth.

Clon1998, 2021-12-23:
Okay, at least now I know it due to the camera widget.
As I said before, I am using a library that provides me the cam widget. I'll have to look into it in order to modify it to ensure that it drops frames on slow connection.

Clon1998, 2022-06-13:
Can I get some feedback on the issue.
In the last few version i added responsive MJPEG and the normal streaming MJPEG that should resolve the issue.

closed 2022-07-03 by Clon1998

Aasikki, 2022-07-03:
> Can I get some feedback on the issue. In the last few version i added responsive MJPEG and the normal streaming MJPEG that should resolve the issue. Especially the responsive MJPEG should drop frames on a slower connection!

Hello, sorry for late reply, haven't been printing for a while so couldn't test it :)

Went to test it today as I got mail from thw issue being closed but now have another issue preventing me from testing 😅 will post as another issue.