Clon1998/mobileraker #35: Kitkat not compatible

pamalofeev, 2022-01-26:
The mobileraker.com website says "*Works on iOS 13.6+, Android Kitkat and above".  However, on my tablet with android 4.4.2, the application cannot be installed, there is a syntax error.  Apkpure determines app compatibility only with android 7.1.

Clon1998, 2022-01-27:
I am not the maintainer of the website. 
Mobileraker definitely never supported Android 4 (SDK 14-19). It started with support for Android 6 (Marshmellow) and currently requires Andorid 7 (SDK 25+) 
See:
https://github.com/Clon1998/mobileraker/blob/92c74903c7b77df6fa32d021a8cc808c635d3e80/android/build.gradle#L3-L6

closed 2022-01-27 by Clon1998

jniebuhr, 2022-01-27:
it's changed on the site

Misterke, 2022-01-28:
It's a pity that support only starts with Nougat as there are still so many Samsung tablets out there running Marshmallow.

Clon1998, 2022-01-28:
Yes for sure.
But I don't have the needs nor do I have the time to make it compatible with an 9 year old android version.
The code is on GitHub, so feel free to adapt it for 4.4.