Clon1998/mobileraker #36: Incorrect Current Layer value

qcgilbert, 2022-02-08:
Current Layer always reported as last layer. E.g. 150/150.

netweaver1970, 2022-03-08:
for me, on iOS, I'm seeing layer 14/15 while Fluidd is showing 16/15 and I'm at my final layer. So it seems both are wrong :)
SuperSlicer tells me there are 15 layers.

Clon1998, 2022-03-08:
> for me, on iOS, I'm seeing layer 14/15 while Fluidd is showing 16/15 and I'm at my final layer. So it seems both are wrong :)
> SuperSlicer tells me there are 15 layers.

I am using the same equation as Fluidd. However the current layer is based on the current Z high.
However I will look into this since I noticed problems with rounding errors in other widgets too.

Clon1998, 2022-03-10:
I did some further investigation related to this issue. As @netweaver1970 pointed out Fluidd for example also has this issue.
For my printer, it's caused due to the fact that I am using bed mesh and therefore the Z value is changed.
Currently, I am printing the first layer at 0.3mm but Z-value is jumping between 0.37 and 0.24mm. Since I am using the current Z value to calculate the current layer high the mesh compensation in my case can cause the wrong layer to be shown in the UI.

closed 2022-03-10 by Clon1998

lucasavendano, 2023-03-25:
Hi @Clon1998 mobileraker is great!, i was looking for an issue like this, and i did find it!😁

Sorry for my may be dumb question but, why using current Z value for calculate current layer?, in my case i use mainsail and the current and total layers shows correctly in browser and also in klipperscreen but in mobileraker this information is always wrong and now i can imagine why. I was thinking that start counting from 0 instead of 1, and as Z value can change due to bed mesh the information is always changing too.

Other thing is M117 messages don´t stay, shows and quickly disappear, in my case are always right, but as the message disappear this reference is lost.

If it is unavoidable use the z value for calculate current layer, can you base your calculation as mainsail do instead of fluidd´s?

Thanks for mobileraker! i wish i had a fraction of your brain🤣