Clon1998/mobileraker #38: Constant Notifications while printing (iOS)

edddeduck, 2022-02-27:
**Describe the bug**
When printing you get notifications every second from the app.

**To Reproduce**
Steps to reproduce the behavior:

1. Open the app then start a print on the printer
2. Notice you start to get notifications every second.

**Expected behaviour**
You would expect to get notifications at key progression points or after set amounts of time.

**Screenshots**
N/A

**Smartphone (please complete the following information):**

This doesn't seem to be device specific but below is the hardware I have been using.

 - Device: iPhone 
 - OS: [e.g. iOS8.1]
 - Browser [e.g. stock browser, safari]
 - Version [e.g. 22]

**Mobileraker-Version
Test Flight 1.4.1 (1.4.1.0)

**Additional context**

- Tagged this as iOS only as I don't have an android device to hand to test.
- Having some options on how and when you get notifications would be useful.
- There is also another notification issue on iOS that also exists and reduces the usefulness of notifications as you currently need to keep the app in the foreground to trigger them. https://github.com/Clon1998/mobileraker/issues/23

natewalck, 2022-02-27:
I am also seeing this as described above. 

iOS 15.3.1

Clon1998, 2022-03-01:
Thanks for the feedback. 
After some investigation, I found out that this is caused by the used notification library. The print progress notification is supposed to be a progress bar/progress notification. However, the library does not support progress bar notifications for iOS yet and therefore creates normal messages. 
Because of that I disabled local notification handling for IOS and removed the progress notification for iOS for now in 79c22fa5f7d6a547f1d8402fe4b540ccd68125ea. However, state update notifications like print start, error, print finished should still be shown.

Clon1998, 2022-03-08:
@jniebuhr submitted a new version to TestFlight that should fix this issue. Please let me know if it works in v 1.4.2

Clon1998, 2022-06-13:
Fixed with 1.6.2

closed 2022-06-13 by Clon1998