Clon1998/mobileraker #39: Works only local

markwinger, 2022-03-01:
I installed this on my phone and it works using local IP address.  But if I connect from outside via port forwarding it doesnt work.  Is there something I need to do to make this work?

Clon1998, 2022-03-01:
Did you install the [Mobileraker Companion](https://github.com/Clon1998/mobileraker_companion)?

markwinger, 2022-03-01:
No.  I was unaware of it.  I'll try when I get back.  Thanks

markwinger, 2022-03-02:
I installed the companion but still nothing.  When I click test is just says Result:not tested.

I added the update to moonraker.conf.  Is there more to setup that I’m missing?

Clon1998, 2022-03-02:
Well companion enables the push notification when you're not in you're local lan. 
However the app requires that the websockt address you enter in the Add/Edit printer page is available in your current network in order to use it. If you want to enable remote access to the print eg. via cellular connection you need to ensure that you can access the printer. This can be done via vpn for example.

markwinger, 2022-03-02:
I have port forwarding on my router on a port.

I can access my camera on the printer by opening:
<wan address>:<port>/webcam/?action=stream

So I am trying to get this to work on my phone.  When I added the printer I entered:
<wan address>:<port>

To get the api key, I open fluid locally on a tablet, and go to the configuration api key.  I then use the camera to read the barcode label moonraker displays.

Moonraker does not work via port forwarding.   I can view the camera remotely, but I can not terminate a print that  has turning into spaghetti,
I requested a feature to allow this more that a year ago but I have heard nothing about progress on this.

When I saw that ap I thought it was the solution for me.  But if I cannot set it up for mobile, there is no point to the ap for me.