Clon1998/mobileraker #40: on IOS, Fetching printer keeps spinning, other functions work

netweaver1970, 2022-03-08:
**Describe the bug**
Fetching printer keeps spinning and nothing comes back. So no control screen
Emergency button works though, also restart FW and restart MCU
Also retrieving files from the printer in the app also works.

**To Reproduce**
add printer on the local network, try to get the main screen

**Expected behavior**
Main control screen appearing 

**Screenshots**
just spinning squarish logo and 'running' text: Fetching printer.

**Desktop (please complete the following information):**
iOS (multiple browsers, Safari, Brave, Chrome)

**Smartphone (please complete the following information):**
 - iPhone X
 - OS: latest, fully updated

**Mobileraker-Version
1.4.1-1.4.1.0

no API key

Clon1998, 2022-03-08:
Hey 👋,
thanks for the bug report. Can you verify that the problem still exists after force closing the app and opening it again?
I should have fixed that bug in aaec8cf68ed3359a1b7533ea7388654e1cf08caf . However the iOS build is a little bit older.

netweaver1970, 2022-03-08:
you're right, the bug is fxed after forced closure. But now I'm getting bombarded with notifications while printing. I saw there was already a bug for this.
Looking good so far! Thanks.

closed 2022-03-08 by Clon1998