Clon1998/mobileraker #43: Add more detailed ETAs and estimates

Charghoul, 2022-04-20:
It would be really great to get all the print time estimates and ETA times that the mainsailOS provides. 
- estimate of klipper 
- estimate of slicer 
- time printed so far
- ETA (maybe explain where the ETA comes from? Add option to choose ETA from klipper estimate or slicer estimate? ) 

Not sure why, but the estimates ETA that mobileraker provides is always different from what mainsailOS shows.

Clon1998, 2022-04-22:
> Not sure why, but the estimates ETA that mobileraker provides is always different from what mainsailOS shows.
The reason for that is that Mobielraker uses only one source while Mainsail combines the mentioned calculation sources and combines them to ensure the best eta. 

Currently Mobileraker uses this formula to calculate the eta:
https://github.com/Clon1998/mobileraker/blob/92c74903c7b77df6fa32d021a8cc808c635d3e80/lib/dto/machine/printer.dart#L35-L42

Clon1998, 2023-12-06:
Eta and estimates are available now via a tooltip by long pressing the ETA cell.

closed 2023-12-06 by Clon1998

locki-cz, 2024-05-03:
There is not possible set ETA to show slicer?

Clon1998, 2024-05-03:
Long pressing the eta cell will bring up the tooltip that shows all available ETAs

[image]

locki-cz, 2024-05-03:
I understand it from previous replyes ... but i can see in mainsail info from slicer on klipper screen info from slicer but there is some weird eta ... is possible add it to setting and see same info also in this great app?
[image]
App now it shows by default AVG ... which is every time very off

Clon1998, 2024-05-04:
> I understand it from previous replyes ... but i can see in mainsail info from slicer on klipper screen info from slicer but there is some weird eta ... is possible add it to setting and see same [...]

I am not quit sure what you are asking for?
The app is able to show all estimates if necessary. 
The ETA that is shown is calculated based of the AVG of all available estimate sources:

https://github.com/Clon1998/mobileraker/blob/9e4e11b67e765d51dde161f10a05f6afb8b2f0bc/common/lib/data/dto/machine/printer.dart#L210-L234

https://github.com/Clon1998/mobileraker/blob/9e4e11b67e765d51dde161f10a05f6afb8b2f0bc/common/lib/data/dto/machine/printer.dart#L181-L185

Are you talking about the ability to select what kind of ETA is shown on the dashboard? So you can choose if you want to use AVG, slicer, file... as source for that?

locki-cz, 2024-05-04:
I just want to see the exact same ETA as displayed in Mainsail or on the klipperscreen. A random average number is useless to me. I don't need to see them all. I just want to see the one that is correct.

I apologize if that wasn't clear from my previous responses.

Clon1998, 2024-05-04:
> I just want to see the exact same ETA as displayed in Mainsail or on the klipperscreen. A random average number is useless to me. I don't need to see them all. I just want to see the one that is correct.
> 
> I apologize if that wasn't clear from my previous responses.

Okay got it.
It is not a random number. In fact I am using the exact same logic as Mainsail does:

https://github.com/mainsail-crew/mainsail/blob/fa61d4ef9297426a404dd845a1a4d5e4525c43dc/src/store/printer/getters.ts#L647-L729

I am currently printing a 12hrs print, and both my mainsail and app show the exact same ETA values.
However, in contrast to Mobileraker, Mainsail lets you pick the ETA calculation sources on which it runs an AVG calculation.