Clon1998/mobileraker #44: Contribution

AntoszHUN, 2022-05-01:
Hi!

Would you like to know if this program is under development? If so, I would be happy to support you in creating the Hungarian language. Is there a need for it?

Clon1998, 2022-05-01:
Hey 👋,
yes the app is on constant development.
However, feel free to add any translation you want. Checkout the dev branch to get the latest language file 😉
Currently only English is fully implemented. There is a file for the german language however i don't have the time to properly add all language keys in German.

Clon1998, 2022-05-01:
Here you can find the english language file.
https://github.com/Clon1998/mobileraker/blob/dev/assets/translations/en.json

AntoszHUN, 2022-05-01:
> 

> Hey 👋, yes the app is on constant development. However, feel free to add any translation you want. Checkout the dev branch to get the latest language file 😉 Currently only English is fully [...]

Thanks for the quick response! Then I do the Hungarian language quickly.

closed 2022-05-09 by AntoszHUN