Clon1998/mobileraker #47: Remote connection not working on tested network

calvin-smith, 2022-05-18:
Hello.  This may be a configuration issue, but something is behaving oddly.

I can send a REST request with X-Api-Key and my API key and everything works fine.  Once I enter the same information into MobileRaker, it gives me an error connecting.  Both of these tests are on the same network that is not local.  I believe that I have permissions and port forwarding is working because I can manually make API calls.

[image]

[image]

Thank you.

jniebuhr, 2022-05-18:
You can make manual requests but you didn't test upgrading them to Websocket (which is what mobileraker is complaining about). Do you use some kind of reverse proxy like nginx in front? You have to configure allowing websockets with that.

Also: just because you edited the post the old URL is not gone. you may want to change that domain now or resort to some kind of VPN to access your printer

calvin-smith, 2022-05-18:
I will do that.

Note: Thanks for the info.  I deleted the revision and will change the url.

closed 2022-05-28 by Clon1998