Clon1998/mobileraker #50: Cancelled print shows as paused

jacrify, 2022-05-28:
**Describe the bug**
A clear and concise description of what the bug is.
After cancelling print, app thinks it is paused. I cannot start another print in app

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

Start a print. 
Cancel it
Look at status ( shows as paused)
Check print in fluid ui, print shows as cancelled. 
Start another print, app picks it up

**Expected behavior**
A clear and concise description of what you expected to happen.
App picks up that print is cancelled

**Screenshots**
If applicable, add screenshots to help explain your problem.
[image]
[image]

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]
iOS, iPad and iPhone

**Smartphone (please complete the following information):**
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - Browser [e.g. stock browser, safari]
 - Version [e.g. 22]

**Mobileraker-Version
This information can be found in Android's and IOS's App-Info.
1.61.1-15

**Additional context**
Add any other context about the problem here.

Clon1998, 2022-07-03:
Works fine on all 3 devices for me.
Can you give me an update if the issue still is valid on your side?

jacrify, 2022-07-03:
It’s working fine Patrick. Sorry it’s a user error

closed 2022-07-03 by Clon1998