Clon1998/mobileraker #51: Allow z offset changes when paused

jacrify, 2022-05-28:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]
I like to baby step when paused. Also between prints
App hides this ui feature when not actively printing. 

**Describe the solution you'd like**
A clear and concise description of what you want to happen.
Turn off hiding please :)

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

Clon1998, 2022-05-28:
Hey,
I'll enable the baby steps while the printer is paused. In the meantime you can go into the app settings an enable to show the baby stepping card always.

jacrify, 2022-05-28:
Ah legendary, I didn’t see that.

Omg there are themes too?

Clon1998, 2022-05-28:
> Ah legendary, I didn’t see that. Omg there are themes too?
> […](#)

Yes!

closed 2022-05-28 by Clon1998