Clon1998/mobileraker #54: (Trivial bug) Enter key on keyboard does not submit console command on ios

jacrify, 2022-06-04:
**Describe the bug**
A clear and concise description of what the bug is.

When hitting done on soft keyboard or enter on physical (bluetooth keyboard) at the end of typing a console command, nothing happens. You need to click the submit arrow (oh the pain, the pain!)

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

Physical keyboard enter key should submit the command
**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]
iOS
**Smartphone (please complete the following information):**
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - Browser [e.g. stock browser, safari]
 - Version [e.g. 22]

**Mobileraker-Version
This information can be found in Android's and IOS's App-Info.
1.6.1-15
**Additional context**
Add any other context about the problem here.

I suspect adding 

`onEditingComplete: model.onCommandSubmit`

to the TextField on line 121 in console_view.dart will fix this, but I'm just guessing.

Clon1998, 2022-06-04:
I guess youre using an IPad with a keyboard.
I am unable to test my implementation against it.
However testing on android emulator seems to suggest that passing enter will now submit the entered command !

closed 2022-06-04 by Clon1998

jacrify, 2022-06-04:
Yeah I originally raised this as I couldn't hit enter on my ipad
(bluetooth) keyboard. But it also happens with the onscreen keyboard.

Thanks for fixing

commit 658ce610 (2022-06-04): Fixed #54
commit 2ac3d87e (2022-06-04): Fixed #54