Clon1998/mobileraker #55: Flickering garbled display

xcasxcursex, 2022-06-07:
**Describe the bug**
Display is flickering and drawing... something strange. See screenshot.

**To Reproduce**
Steps to reproduce the behavior:
1. Add a printer

**Screenshots**
This shot is a moment in time, the re-arrangement happens constantly.
[image]

**Smartphone (please complete the following information):**
 - Device: Galaxy S5
 - OS: lineageos 18.1 (android 11)
 - Version: 18.1 (android 11)
 - Note: No google services installed

**Mobileraker-Version
1.62

**Additional context**
Installed directly from release section APK.

Clon1998, 2022-06-13:
Sadly i can not confirm that on my device nor on my emulator.
My guess is that the issue is related to hardware incompatibility and flutter.
Sadly nothing I can do for you to resolve it.

closed 2022-06-13 by Clon1998

xcasxcursex, 2022-06-13:
I found the cause of this, the app wanted permissions to my camera and was denied. Granting camera permissions made the graphical glitches stop.

xcasxcursex, 2023-01-22:
After coming across this several times now in various apps, today I tracked down the relevant issue, so I'll link it here for your info: 

https://github.com/flutter/flutter/issues/105093