Clon1998/mobileraker #57: Config files access denied

robonxt, 2022-06-17:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'Files' tab
2. Tap on 'Configs'
3. Tap on any '.cfg' file
4. Editor shows '...error code 401 unauthorized...'
5. Tap save button
6. Error banner pops up with 'cannot save file'

**Expected behavior**
Editor should load the file and be able to make changes to them. Saving should work without errors.

**Screenshots**
[image]
[image]

**Smartphone (please complete the following information):**
 - Device: OnePlus 8T
 - OS: Android 11

**Mobileraker-Version
 - 1.6.2 (latest beta from Google Play Store)

Clon1998, 2022-07-03:
Sorry for my late response.
It seems like moonraker does not have the permission to access the config files.
Ensure the file manager is configured correctly.

See
https://moonraker.readthedocs.io/en/latest/configuration/#file_manager

Clon1998, 2022-07-03:
Alternatively the authorization might be configured wrongly.

See
https://moonraker.readthedocs.io/configuration/

robonxt, 2022-07-03:
Weird. I've followed (or at least copied) many of the config setups, so it should work. Here's my condensed moonraker.conf file:

```python
# moonraker.conf
[server]
host: 0.0.0.0
port: 7125

[file_manager]
config_path: ~/klipper_config
log_path: ~/klipper_logs
enable_object_processing: True

[data_store]
temperature_store_size: 6000
gcode_store_size: 1000

[authorization]
force_logins: True
cors_domains:
  *.local
  *.lan
  *://app.fluidd.xyz

trusted_clients:
  10.0.0.0/8
  127.0.0.0/8
  169.254.0.0/16
  172.16.0.0/12
  192.168.0.0/16
  192.168.2.59
  FE80::/10
  ::1/128

# enables partial support of Octoprint API
[octoprint_compat]

# enables moonraker to track and store print history.
[history]

# Timelapse features here
...

# this enables moonraker's update manager
[update_manager]
enable_auto_refresh: True
channel: beta

# this enables fluidd updates
[update_manager client fluidd]
type: web
repo: fluidd-core/fluidd
path: ~/fluidd

# Timelapse update stuff here
...

[update_manager mobileraker]
type: git_repo
path: ~/mobileraker_companion
origin: https://github.com/Clon1998/mobileraker_companion.git
primary_branch:main
managed_services: mobileraker
env: ~/mobileraker-env/bin/python
requirements: scripts/mobileraker-requirements.txt
install_script: scripts/install-mobileraker-companion.sh

# KlipperScreen update stuff here
...
```

What did I missed or misconfigured?

Clon1998, 2022-07-05:
Sadly I am not an expert with moonraker and its config.
However, does opening the config file work on any webinterface?

robonxt, 2022-07-05:
Yes it does. I use fluidd, and the web file manager works normally.

Clon1998, 2022-08-08:
Hey due to other issues related to printer-url parsing.
Can you verify that the `Printer-Address` in the Printer-Settings is correct?
For example, my printer runs on `192.168.178.135` therefore, the correct address should be `http://192.168.178.135`.

robonxt, 2022-09-06:
Sorry for the late response. Yes, the printer address is set properly in the app, if that is what you meant by `Printer-Address`

EricThunder, 2023-08-05:
Have the exact same issue. Really frustrating since everything else in the app works as expected.

Clon1998, 2023-08-06:
> Have the exact same issue. Really frustrating since everything else in the app works as expected.

Could you please post a screenshot of the printer configuration?
Is most cases this is caused by a faulty configured http printer address.

EricThunder, 2023-08-06:
I set-up my printer behind a reverse proxy (nginx proxy manager) so that I can remote anywhere and have a ssl certificat. [image]

Clon1998, 2023-08-06:
> I set-up my printer behind a reverse proxy (nginx proxy manager) so that I can remote anywhere and have a ssl certificat. [image]

I don't think the reverse proxy is the issue. After checking the relevant code section again I seem to have found the reason for the error. I simply forgot to correctly setup the HTTP call that is responsible for downloading any config file from moonraker. Therefore, Can you confirm that the app is showing a 400 status code instead of a 404?

EricThunder, 2023-08-06:
Nope, error code is 401, unauthorized. Maybe I just missed some configuration? But again, everything else works as expected. [image]

Clon1998, 2023-08-06:
> Nope, error code is 401, unauthorized. Maybe I just missed some configuration? But again, everything else works as expected. [image]

No that's perfect!
Its quite late and I confused 400 - Bad Request with 401 - Unauthorized.

This confirms my assumption that I simply forgot to include the header / ApiKey in the request that downloads the file.
I will fix that in the next release. Sadly, there is no temporary workaround for this :(

EricThunder, 2023-08-06:
Well at least now I know its just a bug and don't have to worry about it. I'll just wait until the next update. No rush!

Clon1998, 2023-08-11:
Added in [2.5.0](https://github.com/Clon1998/mobileraker/blob/master/docs/changelog.md#250---2023-08-11)

closed 2023-08-11 by Clon1998

EricThunder, 2023-08-14:
Finally got the update on the app store (2.5.1) but still got the same exact issue.

Clon1998, 2023-08-14:
> Finally got the update on the app store (2.5.1) but still got the same exact issue.

Still 401?

EricThunder, 2023-08-14:
Yup

Clon1998, 2023-08-14:
Could you send me the logs of the app after you once more tried to open the file. The logs can be found at the bottom of the apps settings

EricThunder, 2023-08-14:
[mobileraker_2023-08-14T11-28-32.log](https://github.com/Clon1998/mobileraker/files/12336814/mobileraker_2023-08-14T11-28-32.log)