Clon1998/mobileraker #59: Doesn't show second extruder status

DeanCording, 2022-07-03:
**Describe the bug**
Mobilerake only shows the status of the first extruder on a multi-extruder printer

When printing from the second extruder, it only display the first extruder temp.

**Expected behavior**
Should should the status of the currently active extruder.

**Smartphone (please complete the following information):**
 - Android 12

**Mobileraker-Version
1.6.2

**Additional context**
Add any other context about the problem here.

Clon1998, 2022-07-03:
Hey,
are you by any chance open to test my implementation (Commit: 5b8fbd3d51ca4bcadc189647aa0e40c65b0f7604).
Github actions should build a new APK : https://github.com/Clon1998/mobileraker/releases/tag/android1.7.0-b24

Please let me know if it works since I don't have a dual extruder printer!

DeanCording, 2022-07-04:
Seems to work ok in landscape mode
[image]

But in portrait mode there is too much space between the cards

[image]

Clon1998, 2022-07-05:
I've decreased the min size for the cards in 4e3a8935eff1ba183508045a292bf3288ea2da2f.
Here you can see a 4.7inch emulator screenshot. I hope that fixes your issue. Feel free to open a new issue if it still is not fixed once I release a new beta build!
[image]

closed 2022-07-05 by Clon1998