Clon1998/mobileraker #60: Camera does not flip when selected in settings

jacrify, 2022-07-03:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error
Go to camera in settings
Choose flip vertical or horizontal
Hit save
Camera does not flip
Same result on both cam modes
**Expected behavior**
A clear and concise description of what you expected to happen.
Image should flip. It does flip in front end (mainsail) when I do the same thing there
**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Smartphone (please complete the following information):**
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - Browser [e.g. stock browser, safari]
 - Version [e.g. 22]
iOS 15.6 iPhone XR
**Mobileraker-Version
This information can be found in Android's and IOS's App-Info.
1.6.2-16

**Additional context**
Add any other context about the problem here.

Clon1998, 2022-07-03:
Ion my iphone it works fine.
However, if you flip the cam in the printer's settings and press save, does it close the edit screen or does it stay in the edit screen?

jacrify, 2022-07-03:
Oh dude I’m so sorry: it’s working on mine too, but the camera is rotated
90 degrees so vertical flips horizontal and horizontal flips vertical! I
was looking for the wrong result.

jacrify, 2022-07-03:
User error : user cannot tell the difference between horizontal and vertical

closed 2022-07-03 by jacrify