Clon1998/mobileraker #62: Status page broken

Leithian, 2022-07-04:
**Describe the bug**
When opening the status page there is only a grey screen shown except for the one-word printer status like "Standby". All other pages are displayed correctly.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to the main screen / Open the app

**Screenshots**
[image]

**Smartphone (please complete the following information):**
 - Device: Pixel 4XL
 - OS: Android 12

**Mobileraker-Version
1.7.0-23

**Additional context**
Also clearing cache, app data or reinstalling doesn't work

Clon1998, 2022-07-04:
Related to #61.
Just triggered a new build for the beta test that should resolve the issue.

closed 2022-07-04 by Clon1998

Leithian, 2022-07-04:
Thanks, sorry that I didn't see the other report, my bad!

commit 18c5216a (2025-01-07): feat: Show spoolman page even if klipper is disconnected

Fix #62