Clon1998/mobileraker #64: Total noob here..

TristenStower, 2022-07-12:
Hey, I know people have successfully set this up but I just cant seem to figure it out. Before you jump I have been looking at your docs and moonraker docs for about 2 hours now and made no progress.

I installed your app and typed up the API key, but I keep getting the error that the connection wasn't upgraded to websocket. 

Could you possibly post a screen record of you setting this up on IOS and the moonraker file?

Clon1998, 2022-07-12:
Hey.
This error is often shown if the API key is missing or wrong.
You can verify that you need to enter the API key by checking if you're moonraker.conf file contains the ’[authorization]’ section with the option ’force_login=True’.
If that is the case you need the API else you won't.

To setup the API key you can use the built in QR scanner by clicking into the API key field and pressingly the QR icon next to the text input.
To open the QR code/API key open up ’fluidd-> settings -> authentication-> api key’.
Let me know if that helps.

ryang3d, 2022-08-07:
Running into the same issue on Android (Pixel 6). I have confirmed that my moonraker.conf file contains the `[authorization]` section with the option `force_login=True`, but entering my API key (even after refreshing it) just results in the error saying the connection wasn't upgraded to websocket.

I also tried another Klipper/Android app (Klipperoid) with the same printer and it works perfectly, and no API key required.

[image]

Looks like the same error with or without the API key in the text field

[image]

Would love to give your app a spin if this can be resolved!

Clon1998, 2022-08-07:
> Running into the same issue on Android (Pixel 6). I have confirmed that my moonraker.conf file contains the `[authorization]` section with the option `force_login=True`, but entering my API key [...]

Are you using an IP or mDNS?
Currently mDNS (.local) is not supported

Clon1998, 2022-08-07:
Also @TristenStower are you using an IP or mDNS?

ryang3d, 2022-08-07:
> Are you using an IP or mDNS? Currently mDNS (.local) is not supported

Just local IP

ryang3d, 2022-08-08:
Not sure if this is where @TristenStower was also going wrong, but I just got it to work with a very simple change. For my first few unsuccessful attempts (previous post) at setting up my printer with Mobileraker, I used "http://" to prefix my printer's IP address in the Printer - Address field (so http://192.168.x.x).

Instead, just now I tried entering the printer's IP without the "http://" and it connected immediately. Hope this helps!

[image]

TristenStower, 2022-08-08:
@ryang3d THIS! I pushed this off to the bottom of my to do list, got the email about this thread and tried it on a whim. BAM thank you so much it worked!!

closed 2022-08-08 by TristenStower

Clon1998, 2022-08-08:
> Not sure if this is where @TristenStower was also going wrong, but I just got it to work with a very simple change. For my first few unsuccessful attempts (previous post) at setting up my printer with Mobileraker, I used "http://" to prefix my printer's IP address in the _Printer - Address_ field (so http://192.168.x.x).
> 
> 
> 
> Instead, just now I **_tried entering the printer's IP without the "http://" prefix, and it connected immediately_**. Hope this helps!
> 
> 
> 
> [image]
> 
> 
> 
> PS thanks for building and sharing your app @Clon1998 !

Greate !
I'll have a look into this since it should work with pasting in the http as well!
Still great that it works for both of you now 😊