Clon1998/mobileraker #66: App is freezing on iOS

jacrify, 2022-07-24:
**Describe the bug**
A clear and concise description of what the bug is.
I run an iPad next to my printer. One of the latest releases of the app (I haven't used the printer in two weeks, it was fine before then, and I can see there are new features added) has caused the app to freeze after some period of time.

By freeze, I mean it is totally unresponsive and needs to be force closed and reopened.

It's pretty consistent. Hard to debug though- can I get a  build with debug logs?

**To Reproduce**
Steps to reproduce the behavior:
leave device running with mobile raker on (I think mobile raker keeps iOS awake?)
Wait 10-20 minutes
App is unresponsive. Cannot scroll, buttons don't work, etc. Switching away and back does not fix.

**Expected behavior**
A clear and concise description of what you expected to happen.
Not that :)

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Smartphone (please complete the following information):**
 - Device: iPad mini 5th gen
 - OS: Ios
 - Version 15.5 

**Mobileraker-Version
This information can be found in Android's and IOS's App-Info.
1.71.1-30

**Additional context**
Add any other context about the problem here.

Clon1998, 2022-07-24:
Thanks for your bug report.
I'll have a look at this behavior more indepth next week.
Just checking with my iphone I noticed that mobileraker does not prevent the screensaver/screen off.
Can you provide me with the information:
Which screen did you have open?
- Fullscreen Webcam
- Dashboard (Which tab?)
...

jacrify, 2022-07-24:
Hmm I'll need to check : Maybe I turned off display lock.

I had the dashboard open, main tab. I'll try and do some tests today to see if it is a particular screen or not (I suspect the webcam).

Clon1998, 2022-07-26:
So far I was able to identify the problem.
It seems like the application is running out of memory due to a memory leak in the MJPEG WebCam streaming widget I am using.
For now it seems like this is not an easy/fast fix :(

jacrify, 2022-07-26:
I'll hide the camera on my ios copy. As long as the camera settings don't
replicate to my iphone (I want to be able to see the camera there) then
that will work fine for me.

Thanks for looking into it.

jacrify, 2022-07-29:
Hmm I've removed the camera from the printer definition and I'm still getting consistent and repeatable freezes on the dashboard. I suspect something weird in the telemetry display maybe?

Clon1998, 2022-07-29:
In case the memory leaks are causing the freezes, what i suspect.
I found another memory leak caused by the plots of the temperature sensors. Since the data amount leaking is way smaller the app should freeze not that fast. However, I already started to debug the issue. I hope I can find a workaround to this since it seems like the library providing the plots is leaking...
I'll keep you updated once I resolved any of the leaks 😉

Clon1998, 2022-07-29:
Managed to fix the leaking of the plots in 5296b06c875d951566fdf52664c3de58ec1a6cb2.
I just triggered a new beta build for Android+Ios if you're interested in testing the new build (1.7.2-31) you can use this [testflight](https://testflight.apple.com/join/h9JwuyR0) link.

jacrify, 2022-08-03:
Sorry mate just got a chance to try this properly now. TestFlight build freezes on same device at same point (no camera).

Clon1998, 2022-08-07:
> Sorry mate just got a chance to try this properly now. TestFlight build freezes on same device at same point (no camera).

Damn...
I am currently completely rewriting the app. Let's see if this resolves the issue 🫠

jacrify, 2023-06-29:
Works now thanks!

closed 2023-06-29 by jacrify

Clon1998, 2023-06-29:
> Works now thanks!

That was unexpected 👀
But thanks for the feedback and I am happy that it is working now!
🚀