Clon1998/mobileraker #67: Failed to connect to a printer

dstrelnikov, 2022-08-08:
**Describe the bug**
I haven't used mobileraker for a while but used fluidd both on my PC and on my smartphone instead. Now I wanted to try mobileraker once again but I'm unable to connect to the printer. The printer is located in the same network (my smartphone's IP is in the trusted clients). I can connect to moonraker via fluidd from my smartphone as well as sent API requests from a browser.

I tried different ways to specify the address, particularly:
- `192.168.178.126:7125`
- `http://192.168.178.126:7125`
- `ws://192.168.178.126:7125`

Also all of them with a slash at the end. The API key is not required in my setup, however I also tried to specify it in the corresponding field, however with no success. I'm happy to provide more information if I missed something.

**Screenshots**
[image]

**Smartphone:**
 - Device: Sony Xperia XZ2
 - OS: Android 9

**Mobileraker-Version**:

**Additional context**
Add any other context about the problem here.

Clon1998, 2022-08-08:
Either define the full path to Moonraker's WebSocket instance or remove the port :).
Moonraker's default endpoint is located at `ws://<IP/HOSTNAME>:7125/websocket` currently the screenshot shows `ws://192.168.178.126:7125` as the resulting WS-URL.
So you can either remove the port or enter the full ws-url.

dstrelnikov, 2022-08-08:
Oh wow, thank you, that was quick! It works with `ws://<IP/HOSTNAME>:7125/websocket`. However, when the port is not specified, mobileraker is trying a random one and it does not work.

closed 2022-08-08 by dstrelnikov

Clon1998, 2022-08-08:
> Oh wow, thank you, that was quick! It works with `ws://<IP/HOSTNAME>:7125/websocket`. However, when the port is not specified, mobileraker is trying a random one and it does not work.

AS far as I remember it without looking into the code it should just fallback to the default port of 7125 of moonraker.