Clon1998/mobileraker #69: Error on start page (refresh printer error)

peppelorum, 2022-08-21:
**Describe the bug**
An error occurs on the start page, it’s dismissible but pops up again and 
See screenshot

[image]

**Smartphone (please complete the following information):**
 - Device: iPhone 13 mini
 - OS: latest

**Mobileraker-Version
1.7.1-30
**Additional context**
Add any other context about the problem here.

peppelorum, 2022-08-21:
Exception:
 type '_OneByteString' is not a subtype of type 'List<dynamic>' in type cast

#0      new ConfigStepper.parse (package:mobileraker/data/dto/config/config_stepper.dart:33)
#1      new ConfigFile.parse.<anonymous closure> (package:mobileraker/data/dto/config/config_file.dart:53)
#2      Iterable.forEach (dart:core/iterable.dart:325)
#3      new ConfigFile.parse (package:mobileraker/data/dto/config/config_file.dart:28)
#4      PrinterService._updateConfigFile (package:mobileraker/service/moonraker/printer_service.dart:568)
#5      PrinterService._parseObjectType (package:mobileraker/service/moonraker/printer_service.dart:348)
#6      PrinterService._parseQueriedObjects.<anonymous closure> (package:mobileraker/service/moonraker/printer_service.dart:392)
#7      _LinkedHashMapMixin.forEach (dart:collection-patch/compact_hash.dart:614)
#8      PrinterService._parseQueriedObjects (package:mobileraker/service/moonraker/printer_service.dart:391)
#9      PrinterService._printerObjectsQuery (package:mobileraker/service/moonraker/printer_service.dart:746)
<asynchronous suspension>
#10     PrinterService.refreshPrinter (package:mobileraker/service/moonraker/printer_service.dart:108)
<asynchronous suspension>

Clon1998, 2022-08-21:
Thanks for your bug report.
It seems like the error thrown while mobileraker parses your config file. If you don't mind, can you provide me your stepper config section?

peppelorum, 2022-08-21:
Hope I got it all!

#####################################################################
# 	X/Y Stepper Settings
#####################################################################

[stepper_x]
##	Connected to X on mcu_xye (B Motor)
step_pin: P2.2
dir_pin: !P2.6
enable_pin: !P2.1
rotation_distance: 40
microsteps: 16
full_steps_per_rotation:400  #set to 400 for 0.9 degree stepper
endstop_pin: !P1.28
position_min: 0
##--------------------------------------------------------------------

##	Uncomment below for 250mm build
#position_endstop: 250
#position_max: 250

##	Uncomment for 300mm build
#position_endstop: 300
#position_max: 300

##	Uncomment for 350mm build
position_endstop: 345
position_max: 345

##--------------------------------------------------------------------
homing_speed: 50   #Max 100
homing_retract_dist: 5
homing_positive_dir: true

##	Make sure to update below for your relevant driver (2208 or 2209)
[tmc2209 stepper_x]
uart_pin: P1.17
interpolate: True
run_current: 0.8
hold_current: 0.7
sense_resistor: 0.110
stealthchop_threshold: 0

[stepper_y]
##	Connected to Y on mcu_xye (A Motor)
step_pin: P0.19
dir_pin: !P0.20
enable_pin: !P2.8
rotation_distance: 40
microsteps: 16
full_steps_per_rotation:400  #set to 400 for 0.9 degree stepper
endstop_pin: !P1.26
position_min: 0
##--------------------------------------------------------------------

##	Uncomment for 250mm build
#position_endstop: 250
#position_max: 250

##	Uncomment for 300mm build
#position_endstop: 300
#position_max: 300

##	Uncomment for 350mm build
position_endstop: 350
position_max: 350

##--------------------------------------------------------------------
homing_speed: 50  #Max 100
homing_retract_dist: 5
homing_positive_dir: true

##	Make sure to update below for your relevant driver (2208 or 2209)
[tmc2209 stepper_y]
uart_pin: P1.15
interpolate: True
run_current: 0.8
hold_current: 0.7
sense_resistor: 0.110
stealthchop_threshold: 0

#####################################################################
# 	Z Stepper Settings
#####################################################################

## Z MCU - In X Position
## Z0 Stepper - Front Left
[stepper_z]
step_pin: z:P2.2
dir_pin: z:P2.6
enable_pin: !z:P2.1
rotation_distance: 40
gear_ratio: 80:16
microsteps: 16
endstop_pin: z:P1.25
##  Z-position of nozzle (in mm) to z-endstop trigger point relative to print surface (Z0)
##  (+) value = endstop above Z0, (-) value = endstop below
##	Increasing position_endstop brings nozzle closer to the bed
##  After you run Z_ENDSTOP_CALIBRATE, position_endstop will be stored at the very end of your config
#position_endstop: -0.5
##--------------------------------------------------------------------

##	Uncomment below for 250mm build
#position_max: 240

##	Uncomment below for 300mm build
#position_max: 290

##	Uncomment below for 350mm build
position_max: 340

##--------------------------------------------------------------------
position_min: -15
homing_speed: 50
second_homing_speed: 3
homing_retract_dist: 3

##	Make sure to update below for your relevant driver (2208 or 2209)
[tmc2209 stepper_z]
uart_pin: z:P1.17
interpolate: true
run_current: 0.8
hold_current: 0.8
sense_resistor: 0.110
stealthchop_threshold: 0

##	Z MCU - In Y Position
##	Z1 Stepper - Rear Left
[stepper_z1]
step_pin: z:P0.19
dir_pin: !z:P0.20
enable_pin: !z:P2.8
rotation_distance: 40
gear_ratio: 80:16
microsteps: 16

##	Make sure to update below for your relevant driver (2208 or 2209)
[tmc2209 stepper_z1]
uart_pin: z:P1.15
interpolate: true
run_current: 0.8
hold_current: 0.80
sense_resistor: 0.110
stealthchop_threshold: 0

##	Z MCU - In Z Position
##	Z2 Stepper - Rear Right
[stepper_z2]
step_pin: z:P0.22
dir_pin: z:P2.11
enable_pin: !z:P0.21
rotation_distance: 40
gear_ratio: 80:16
microsteps: 16

##	Make sure to update below for your relevant driver (2208 or 2209)
[tmc2209 stepper_z2]
uart_pin: z:P1.10
interpolate: true
run_current: 0.8
hold_current: 0.80
sense_resistor: 0.110
stealthchop_threshold: 0

##	Z MCU - In E0 Position
##	Z3 Stepper - Front Right
[stepper_z3]
step_pin: z:P2.13
dir_pin: !z:P0.11
enable_pin: !z:P2.12
rotation_distance: 40
gear_ratio: 80:16
microsteps: 16

##	Make sure to update below for your relevant driver (2208 or 2209)
[tmc2209 stepper_z3]
uart_pin: z:P1.8
interpolate: true
run_current: 0.8
hold_current: 0.80
sense_resistor: 0.110
stealthchop_threshold: 0

Clon1998, 2022-08-21:
Thanks.
Just briefly looking at the config everything seems to be perfectly normal.
I'll have to look deeper into that later.

closed 2022-10-04 by Clon1998