Clon1998/mobileraker #70: Enhance 'Output pin' behaviour

willkaxxx, 2022-08-21:
* Currently for output button we have got slider, even if it have only two positions
[image]

* Could you please add ordinary button (on /off) , if output pin can take only two values?

* Example from Fluidd
[image]

Thanks!)

Clon1998, 2022-08-21:
Adding such an option should not be a problem. I'll add it once I am done migrating to a new framework.

Fl0ken, 2023-04-25:
Adding to this: could the slider actually enable values between 0 and 1? I got a Daylight on the pin and can only do off/max brightness instead of changing the brightness.

Clon1998, 2023-05-19:
> Adding to this: could the slider actually enable values between 0 and 1? I got a Daylight on the pin and can only do off/max brightness instead of changing the brightness.

Hey, this depends on the output pin `scale` configuration. Make sure you set the configuration correctly to enable this!

Clon1998, 2023-05-19:
@willkaxxx sorry for my WAY to late reply, I just went trough the issues and was wondering why this ticket is not closed yet since I implemented the logic for this ages ago. Turns out the config parsing for the `output_pin` was not working correctly causing the app to still use a slider instead of the new switch cards!

I included the fix in the upcoming version `2.3.0`!

Clon1998, 2023-05-19:
Added in [2.3.0](https://github.com/Clon1998/mobileraker/releases/tag/android2.3.0)

Fl0ken, 2023-05-20:
> > Adding to this: could the slider actually enable values between 0 and 1? I got a Daylight on the pin and can only do off/max brightness instead of changing the brightness.
> 
> 
> 
> Hey, this depends on the output pin `scale` configuration. Make sure you set the configuration correctly to enable this!

Hey, will try the scale when home, but shouldn't this also work if I got PWM enabled? I can already change the brightness in Mainsail.

willkaxxx, 2023-05-21:
Hey, @Clon1998 , I have tried to reinstal app, clean cashes, delete printer from configs.
But it still a slider :(
Could you suggest something to try more?
My current version is 2.3.1
My config in klipper like: 
[image]

Thanks

Clon1998, 2023-05-21:
> > > Adding to this: could the slider actually enable values between 0 and 1? I got a Daylight on the pin and can only do off/max brightness instead of changing the brightness.
> > 
> > 
> > 
> > Hey, this depends on the output pin `scale` configuration. Make sure you set the configuration correctly to enable this!
> 
> Hey, will try ```scale``` when home, but shouldn't this also work if I got PWM enabled? I can already change the brightness in Mainsail.

Good point, I will make sure that non pwm pins are always toggles/switches.

Clon1998, 2023-05-21:
> Hey, @Clon1998 , I have tried to reinstal app, clean cashes, delete printer from configs.
> 
> But it still a slider :(
> 
> Could you suggest something to try more?
> 
> My current version is 2.3.1
> 
> My config in klipper like: 
> 
> [image]
> 
> 
> 
> Thanks

Hey,
I will check that. The pin I tested that with is defined like that
[image]

Clon1998, 2023-05-26:
@willkaxxx, I just had another look and I simply can not reproduce this on my device (version 2.3.4).

See my test config:
[image]

willkaxxx, 2023-05-27:
@Clon1998 I have found the root cause, if name of pin is "LED" (upper case), it doesnt work as expected, but if I change naming to "led" (lower case), it works fine

Clon1998, 2023-05-27:
> @Clon1998 I have found the root cause, if name of pin is "LED" (upper case), it doesnt work as expected, but if I change naming to "led" (lower case), it works fine

That's interesting and definitely a Bug in mobileraker. I will double check my implementation.
Thanks for your reply!

Clon1998, 2023-12-06:
I forgot to close this issue. The parsing for most klipper objects was reworked and the keys/config keys are now case-insensitive.