Clon1998/mobileraker #73: Error when adding https printer

wociscz, 2022-09-24:
**Description**
A clear and concise description of what the bug is.
When trying to add printer behind https I'm getting error
`HandshakeException: Handshake error in client (OS Error: CERTIFICATE_VERIFY_FAILED: unable to get local issuer certificate(handshake.cc:393))`

I have my own certification authority for internal purposes, and imported root/intermediate CA into the phone trust store itself. 3d printer web ui works via mobile chrome browser without an error (certificate is valid).

**To Reproduce**
1. Have the https fluidd/klipper endpoint (this is out of the scope of this issue)
2. Try to add https:// printer in the app
3. Press Test button
4. See error

**Expected behavior**
Connect to https:// printer without problem

**Screenshots**
[image]

**Smartphone**
 - Device: Redmi Note 9T
 - OS: Android12 (Latest miui 13 22.9.21)
 - Browser: Chrome
 - Version: 105.0.5195.136

**Mobileraker-Version**
1.7.1

Clon1998, 2022-09-24:
Hey,
I will look into this. Not to sure why flutter isn't using the internal Trust store. If i am unable to get it working i guess I'll just add an option to add an p12 or similar as truststore directly.

wociscz, 2022-09-24:
It might be sufficient to have checkbox like "allow untrusted" or similar some `--no-check` option (as I noted, am not developer, don't know what options are available).
Thanks!

closed 2022-09-29 by Clon1998

wociscz, 2022-10-11:
Sorry to say that, but still the same error (Trust self-signed certificates checked)

commit 5b2c96fe (2022-09-29): Closes #73