Clon1998/mobileraker #77: Specify input parameters for klipper macros

dummylabs, 2022-10-11:
**Is your feature request related to a problem? Please describe.**
I've got a bunch of parametrised klipper macros which I'd like to run from within the app. It doesn't make much sense to run them with default values.

**Describe the solution you'd like**
It would be great if mobileraker is able to run macro with parameters specified by user, like fluidd does. E.g. when user long press a macro badge, the dialog appears.

**Additional context**
Attached is an example of parameters dialog for pressure advance calibration macro from fluidd.

[image]

Clon1998, 2022-10-14:
I will look for a good way to add this feature. Probably by opening a dialog to enter the parameters.

Clon1998, 2022-12-18:
Added in f6181efb25b244715bb939fe2b5c1c2e5d7330ed

Clon1998, 2023-02-13:
Available soon in version 2.1.x

closed 2023-02-13 by Clon1998