Clon1998/mobileraker #78: Error after upgrading

DABKLUG, 2022-10-11:
**Describe the bug**
Exception:
 type 'Null' is not a subtype of type 'List<dynamic>' in type cast

#0      new ConfigStepper.parse (package:mobileraker/data/dto/config/config_stepper.dart:33)
#1      new ConfigFile.parse (package:mobileraker/data/dto/config/config_file.dart:57)
#2      PrinterService._updateConfigFile (package:mobileraker/service/moonraker/printer_service.dart:757)
#3      PrinterService._parseObjectType (package:mobileraker/service/moonraker/printer_service.dart:426)
#4      PrinterService._temperatureStore.<anonymous closure> (package:mobileraker/service/moonraker/printer_service.dart:377)
#5      _LinkedHashMapMixin.forEach (dart:collection-patch/compact_hash.dart:617)
#6      PrinterService._parseQueriedObjects (package:mobileraker/service/moonraker/printer_service.dart:496)
#7      PrinterService._printerObjectsQuery (package:mobileraker/service/moonraker/printer_service.dart:1032)
<asynchronous suspension>
#8      PrinterService.refreshPrinter (package:mobileraker/service/moonraker/printer_service.dart:168)
<asynchronous suspension>

**Mobileraker-Version
2.0.0-71

**Additional context**
Error makes is impossible to use the app...

Clon1998, 2022-10-11:
Can you provide me your entire config file for your steppers. It seems like there is an issue with the parsing.

DABKLUG, 2022-10-11:
Hi Patrick

Thank you for your quick response.
It's strange, because I have no error in fluid/mainsail?!
Before upgrading Mobileraker showed an error, but it worked anyway. Now
it's totally messed up.

*PART OF PRINTER.CFG*
[stepper_x]
step_pin: ar54
dir_pin: ar55
enable_pin: !ar38
rotation_distance: 40
microsteps: 16
full_steps_per_rotation: 200
endstop_pin: ^ar3
position_endstop: 0
position_min: 0
position_max: 320
homing_speed: 50
homing_retract_dist: 5

[stepper_y]
step_pin: ar60
dir_pin: !ar61
enable_pin: !ar56
rotation_distance: 40
microsteps: 16
full_steps_per_rotation: 200
endstop_pin: ^ar14
position_endstop: 0
position_min: 0
position_max: 315
homing_speed: 50
homing_retract_dist: 5

[stepper_z]
step_pin: ar46
dir_pin: !ar48
enable_pin: !ar62
rotation_distance: 8
microsteps: 16
full_steps_per_rotation: 200
endstop_pin: probe:z_virtual_endstop
position_min: -3
position_max: 395

Best regards
David A. Brandt Klug

Clon1998, 2022-10-11:
> Hi Patrick
>
> Thank you for your quick response.
> It's strange, because I have no error in fluid/mainsail?!
> Before upgrading Mobileraker showed an error, but it worked anyway. Now
> it's totally messed [...]

Thanks. I think the problem is that the app expects a gear ratio. I will push a hot fix tomorrow that should resolve that issue.

Clon1998, 2022-10-12:
Should be fixed in [46b447f](https://github.com/Clon1998/mobileraker/commit/46b447fdec6e3366bdfdd8901b7fa2aa4607b828). New IOS+Android versions are currently building/in processing.

DABKLUG, 2022-10-12:
Awesome, thanks!

Best regards
David A. Brandt Klug