Clon1998/mobileraker #80: Refresh Printer Error, Could not parse type 'Null' is not a subtype of type 'String'

Avatarsia, 2022-10-15:
**Describe the bug**
I have a voron 2.4 r1 with the mainboard recore a6
i cant connect it with mobile raker, normal website works fine.
The connection Test is Sucessfull but after Save the Data it Shows the Topic Error with following subtext:

Exception:
 type 'Null' is not a subtype of type 'String'

#0      new ConfigStepper.parse (package:mobileraker/data/dto/config/config_stepper.dart:29)
#1      new ConfigFile.parse (package:mobileraker/data/dto/config/config_file.dart:57)
#2      PrinterService._updateConfigFile (package:mobileraker/service/moonraker/printer_service.dart:757)
#3      PrinterService._parseObjectType (package:mobileraker/service/moonraker/printer_service.dart:426)
#4      PrinterService._temperatureStore.<anonymous closure> (package:mobileraker/service/moonraker/printer_service.dart:377)
#5      _LinkedHashMapMixin.forEach (dart:collection-patch/compact_hash.dart:617)
#6      PrinterService._parseQueriedObjects (package:mobileraker/service/moonraker/printer_service.dart:496)
#7      PrinterService._printerObjectsQuery (package:mobileraker/service/moonraker/printer_service.dart:1032)
<asynchronous suspension>
#8      PrinterService.refreshPrinter (package:mobileraker/service/moonraker/printer_service.dart:168)
<asynchronous suspension>

**Screenshots**
on request if needed

**Smartphone (please complete the following information):**
 - Device: One PLus 5T
 - OS: Android 12

**Mobileraker-Version
 - Version 2.0.0-71

**Additional context**
Already connected Printers wont have this issue

Clon1998, 2022-10-15:
Can you move to the latest version 2.0.1-74 by upgrading via the Playstore?
It should fix the issue you encounter. Alternatively you should ensure that the enable pin is present in all your [stepper_<NAME>] configs.

Avatarsia, 2022-10-15:
The Update is not yet possible, the last shown update was on the 4.10.2022 in the google playstore.
the recore a6 have integrated drivers so the enable pin is not needed inside the configuration
but i can wait until a fix is online

Clon1998, 2022-10-15:
> The Update is not yet possible, the last shown update was on the 4.10.2022 in the google playstore. the recore a6 have integrated drivers so the enable pin is not needed inside the configuration but i can wait until a fix is online

It seems like I forgot to actually press release for Google Play. I just started the rollout for 2.0.1. Still, it might take a while until it is available in all regions.

Avatarsia, 2022-10-15:
Timer Started :D

Avatarsia, 2022-10-15:
yes update is rolled out and issue is fixed, perfect !

closed 2022-10-15 by Avatarsia