Clon1998/mobileraker #81: Notification Customizations

grahamgraham, 2022-10-24:
I have a feature request. 

It would be great if we could customize the information when we get a notification, and possibly with a snapshot so it can be viewed in the Notification Center easily. 

Currently it says

```
 “Print progress of Ender 3”
ETA:xx:xx 25%
```

If we could change what that says, that would be awesome. Either in a config file or in the app. 

For instance:
```
“Print progress of <print name> on <printer name>”
<time options>, <percentage>
```
Time options could be ETA, time remaining, etc.

Clon1998, 2022-10-27:
That's actually a brilliant idea.
I will look into that since I am rewriting the entire notification implementation anyway including Backend, companion and app integration to fix problems on iOs 🙂.

Clon1998, 2023-03-21:
This was added via the new mobileraker_companion.

You can either define custom M117 Notifications or customize the format directly by editing this [file](https://github.com/Clon1998/mobileraker_companion/blob/main/i18n.py).