Clon1998/mobileraker #84: WearOS support

solo2424, 2022-11-06:
**Is your feature request related to a problem? Please describe.**
Would like to be able to monitor and control printer from WearOS watch. Similar to OctoPrint App (https://play.google.com/store/apps/details?id=de.crysxd.octoapp&hl=en_US&gl=US)

**Describe the solution you'd like**
Add WearOS support for your app

**Describe alternatives you've considered**
I am not a programmer so I have not alternate solutions to accomplish this.

**Additional context**
Even if controls are limited, being able to look at your watch to see the status of a print would be very useful.

Clon1998, 2022-11-06:
Sadly I don't own any wear OS device nor do I have the time to work on such an implementation rn.
Maybe there is another dev who wants such a feature and has the time and motivation to work on it :). I will always happily merge useful PRs to my project.

solo2424, 2022-11-06:
Ok, I may be willing to pay a dev to implement such a feature if there are any out there and the price is reasonable. If there are any devs out there interested me, reach out to me.

perivarlura, 2022-12-06:
should be a matter of dpi scaling mostly, as the wearos allows for side loading of phone apks. not that I am trying to change your decision, and I am not an Android developer, but a wearos scaled apk would be awesome if you change your mind 🙂