Clon1998/mobileraker #87: No GCodes

Actionralf, 2022-11-26:
GCodes are not showing in the files tab.

I can‘t see any GCodes and can‘t start out of the app.

[image]

 - Device: iPad Pro 11“ und iPhone 13 Pro
 - OS: 16.1

Clon1998, 2022-11-26:
Hey!
Can you visit the following URL, with adjusted parameters, and post the result of it here?:

GET http://<PRINTER-IP Address>/server/files/list?root=gcodes

Actionralf, 2022-11-26:
Sure, here is the result:

{"result": [{"path": "Pencil_holder_2_PLA_50m.gco", "modified": 1669484278.5545475, "size": 845058, "permissions": "rw"}, {"path": "Pencil_holder_mid_PLA_1h59m.gco", "modified": 1669486357.1638558, "size": 949534, "permissions": "rw"}, {"path": "Pencil_holder_mid_PLA_2h1m.gco", "modified": 1669498374.2762265, "size": 1004175, "permissions": "rw"}, {"path": "test_Pencil_holder_mid_PLA_2h1m.gco", "modified": 1669493501.322561, "size": 1004175, "permissions": "rw"}, {"path": "xyzCalibration_cube_PLA_25m.gco", "modified": 1669480774.454252, "size": 819094, "permissions": "rw"}, {"path": "xyzCalibration_cube_PLA_30m.gco", "modified": 1669406161.7775545, "size": 845525, "permissions": "rw"}]}

Clon1998, 2022-11-26:
Thanks!
could you try to rename one of the files as `.gcode` instead of `.gco`?

Actionralf, 2022-11-26:
Great! Now i can see the gcode. Now i have to figure out how i can get PrusaSlicer to put out gcode as .gcode.
Thank you very much!

Clon1998, 2022-11-26:
> Great! Now i can see the gcode. Now i have to figure out how i can get PrusaSlicer to put out gcode as .gcode. Thank you very much!

Awesome! However, thanks to you I will also add `.gco` as a supported file type. But this will take some time until its available in the new version :)

closed 2022-11-26 by Clon1998