Clon1998/mobileraker #90: Suggestion on pause/resume/cancel interface (cancel without pausing first)

dstrelnikov, 2022-12-09:
At the moment there is a pause button during the print which then turns to resume/cancel options once pressed.

I suggest turning the pause button into a menu button which then turns to pause/cancel or resume/cancel once pressed. The style could be like now for the SBC power options or something more like material style with two circles.

The reason for my suggestion is the following. Even if I press pause and then cancel quickly, the `PAUSE` macro is still triggered before the `CANCEL_PRINT` macro. This can create additional mess when one wants to quickly stop the print. For example, my `PAUSE` is parking the print head in the front of the build plate (for easy filament swap) while `CANCEL_PRINT` parks it in the back.

It is not a feature request but an invitation for a discussion.

Clon1998, 2022-12-09:
Hey there 👋🏻 ,
tbh. I just implemented it how I liked it back then and how my workflow always is. Replacing the pause button with a FAB that shows multiple options (Pause, Cancel) might event prevent clicking the pause btn by mistake.
I am not to sure what you mean by " SBC power options"

dstrelnikov, 2022-12-09:
> I am not to sure what you mean by " SBC power options"

I meant a single board computer, most commonly a Raspberry Pi. While the printer is idle, this button opens a menu where one can shutdown or reboot it (sorry, I'm in a middle of a long print, cannot check the exact formulations right now).

The button which is currently `PAUSE` could open a similar menu.

Clon1998, 2022-12-10:
> > I am not to sure what you mean by " SBC power options"
> 
> I meant a single board computer, most commonly Raspberry Pi. While the printer is idle, this button opens a menu where one can shutdown or reboot it (sorry, I'm in a middle of a long print, cannot check the exact formulations right now).
> 
> The button which is currently `PAUSE` could open a similar menu.

Got it. You are talking about the bottomsheet if the machine is idling. E.g  (Also currently running a print so I can not take a screenshot from the app directly):
[image]

It is a valid option. For me, a bottomsheet might be a bit overkill. I think I will stick to the FloatingActionButton that expands (SpeedDial) just like it does if the printer is in the pause state.

dstrelnikov, 2022-12-10:
> I think I will stick to the FloatingActionButton that expands (SpeedDial) just like it does if the printer is in the pause state or like on the cfg edit page

I actually like it even more.

Clon1998, 2022-12-18:
Added via 
d4f7e680fb46ea7f5f69c54dfb742b80a980a4a1
d5e61a0a8ddf0ab49afcefb1280fa4312599bcdb

dstrelnikov, 2023-01-12:
Sorry, just came back from holidays. I've tested it, works great! Thanks a lot!

closed 2023-01-12 by dstrelnikov