Clon1998/mobileraker #93: Don't get any Notifications

Experte9, 2022-12-28:
**Describe the bug**
I followed the instructions for Mobileraker's Companion. Also added a Mobileraker.conf because I got 2 Printers on my Pi. 
Then I did a reboot. 
I'm in the App and keep it open in Background. 

**Expected behavior**
A Notification on 25, 50, 75 and completed Print. 

**Screenshots**
No notification, no screenshots 😅

**Smartphone:**

 - Device: iPhone 12 Pro Max
 - OS: iOS 16.2
 - App Version 2.0.2-76

Maybe I missed something, sorry if so...

Clon1998, 2022-12-28:
Can you please provide me with your Mobileraker.log located in `/tmp/`

Experte9, 2022-12-28:
Yeah, sure.
[mobileraker.log](https://github.com/Clon1998/mobileraker/files/10316104/mobileraker.log)

_Would it be okay if we continue in German? I just notice that u maybe from Germany._

Clon1998, 2022-12-28:
> Yeah, sure. [mobileraker.log](https://github.com/Clon1998/mobileraker/files/10316104/mobileraker.log)
> 
> Also, here my .conf (of course its a .conf but I cant upload a .conf here) [Mobileraker.txt](https://github.com/Clon1998/mobileraker/files/10316123/Mobileraker.txt)
> 
> _Would it be okay if we continue in German? I just notice that u maybe from Germany._

Ja sicher 😄.
In der conf Datei von dir hast du zwei Drucker drinnen. Allerdings ließt der Companion nur einen aus der Datei.
Bist du sicher, dass die richtige Config genommen wird ?
Die, die der Drucker ließt liegt unter ` Found configuration file at: /home/pi/klipper_config/mobileraker.conf`

Experte9, 2022-12-29:
Also unter dem Pad von dir liegt keine _mobileraker.conf_
Die Daten von `printer.conf`, `moonraker.conf` usw. liegt unter `/home/pi/printer_1_data/config` und `/home/pi/printer_2_data/config`
Die aktuelle _mobileraker.conf_ liegt unter `/home/pi/mobileraker_companion`

Was müsste ich jetzt wohin schieben damit es funktioniert? Wahrscheinlich die _mobileraker.conf_ in den von dir angegebenen Pad?

Clon1998, 2022-12-29:
> Also unter dem Pad von dir liegt keine _mobileraker.conf_ Die Daten von `printer.conf`, `moonraker.conf` usw. liegt unter `/home/pi/printer_1_data/config` und `/home/pi/printer_2_data/config` Die [...]

Komisch, der Log sagt eindeutig, dass eine Datei in diesem Pfad existiert... 

Ansonsten kannst du diese Sachen ausprobieren:
- Sicherstellen, dass die Datei `Mobileraker.conf` und nicht `mobileraker.conf` heiß?
- Deine `Mobileraker.conf` zum Pfad `/home/pi/Mobileraker.conf` schieben

Experte9, 2022-12-29:
So, ich habe mal Bilder vom Verzeichnis vor und nach dem verschieben gemacht.
[image]
[image]
[image]
[image]
[image]

Allerdings habe ich jetzt das Problem das ich den 2. Drucker nicht mehr erreiche...
außerdem kann ich [remove-mobileraker-companion.sh] nicht ausführen weil mir rechte fehlen.

Clon1998, 2022-12-29:
```log
2022-12-29 18:32:13,412 [mobileraker.py:get_config_file_location()] - Found configuration file at: /home/pi/Mobileraker.conf
2022-12-29 18:32:13,414 [mobileraker.py:__init__()] - Read 2 printer config sections
2022-12-29 18:32:13,419 [mobileraker.py:info()] - [Ender links] Trying to connect to: ws://192.168.178.66:7125/websocket api key <NO API KEY>
2022-12-29 18:32:13,526 [mobileraker.py:info()] - [Ender rechts] Trying to connect to: ws://192.168.178.66:7126/websocket api key <NO API KEY>
```
Also jetzt wird die richtige Config gelesen. D.h. du solltest die Config in `/home/pi/klipper_config` löschen um nicht die falsche zu editieren.

Warum einer der Zwei Drucker jetzt nicht mehr erreichbar ist, kann ich dir nicht sagen. Der Companion ist auf jeden Fall mit beiden verbunden:

```
22-12-29 18:44:00,331 [mobileraker.py:info()] - [Ender rechts] Fetching printer Objects Try#0
2022-12-29 18:44:00,334 [mobileraker.py:info()] - [Ender rechts] Received Server Info
2022-12-29 18:44:00,335 [mobileraker.py:info()] - [Ender rechts] Querying printer Objects
2022-12-29 18:44:00,572 [mobileraker.py:info()] - [Ender rechts] Subscribing to printer Objects
2022-12-29 18:44:01,832 [mobileraker.py:info()] - [Ender rechts] Fetching printer Objects Try#1
2022-12-29 18:44:01,835 [mobileraker.py:info()] - [Ender rechts] Received Server Info
2022-12-29 18:44:01,835 [mobileraker.py:info()] - [Ender rechts] Querying printer Objects
2022-12-29 18:44:02,080 [mobileraker.py:info()] - [Ender rechts] Subscribing to printer Objects
2022-12-29 18:44:03,076 [mobileraker.py:info()] - [Ender links] WebSocket connected
2022-12-29 18:44:03,077 [mobileraker.py:info()] - [Ender links] Fetching printer Objects Try#0
2022-12-29 18:44:03,084 [mobileraker.py:info()] - [Ender links] Received Server Info
2022-12-29 18:44:03,085 [mobileraker.py:info()] - [Ender links] Querying printer Objects
2022-12-29 18:44:03,316 [mobileraker.py:info()] - [Ender links] Subscribing to printer Objects
```

Experte9, 2022-12-29:
Soooo, ich erreiche den 2. Drucker wieder. Der Fehler lag bei mir. Irgendwie hatte es mir die "Remotedrucker" aus den Einstellungen gehauen.

Ich würde dir noch mal das Log anhängen. Kannst du mal schauen ob alles passt?
[mobileraker.log](https://github.com/Clon1998/mobileraker/files/10321362/mobileraker.log)
 Ansonsten starte ich jetzt mal ein druck und schaue ob ich Push bekomme.

Clon1998, 2022-12-29:
> Soooo, ich erreiche den 2. Drucker wieder. Der Fehler lag bei mir. Irgendwie hatte es mir die "Remotedrucker" aus den Einstellungen gehauen.
>
> Ich würde dir noch mal das Log anhängen. Kannst du mal [...]

Ja geht, und dein Handy ist auch bereits ready für Pushmeldungen

Clon1998, 2023-01-07:
Kannst du mir ein Update geben ob die Push-Meldungen jetzt gehen?

Experte9, 2023-01-07:
Oh, tut mir leid. 
Ja es funktioniert jetzt super. Man muss halt die App im Hintergrund offen lassen, aber das weißt du ja und steht auch so beschrieben.
[image]
[image]

closed 2023-01-07 by Experte9

Experte9, 2023-01-12:
> I am having a similar problem with mainsailos (pi image). And wondering that what was the solution here?

The problem was that my config was not in the correct directory. You can move your` Mobileraker.conf` to the path `/home/pi/Mobileraker.conf`

Experte9, 2023-07-29:
Hi again, 

I run into Problems again:(
Over the days I got less updates and now I don't get any Updates. 
Maybe it's easy but I don't know where to start. 
What you need to help?