Clon1998/mobileraker #94: Unable to add printer on iPadOS due to local network access

chriswhite72, 2023-01-03:
**Describe the bug**
When I try to add a printer on my iPad, the connectivity test fails.  This works fine on my iPhone, and after verifying the IP address multiple times, I discovered that mobileraker is properly requesting **local network** access (Settings/Privacy/Local Network) on iPhone but not on iPad.  I deleted and re-installed this application on my iPad but I never get a prompt for **local network** access and the application is not listed under the **local network** settings on iPad.

**To Reproduce**
Steps to reproduce the behavior:
1. Install mobileraker on an iPad (16.2)
2. Click on 'add printer
3. Enter IP address 
4. Click the TEST button.
5. Connection test will fail

**Expected behavior**
iPadOS should prompt the user for local network app permission.  Without this permission, the application cannot access a printer on the local network.
The mobileraker application should be listed under Settings/Privacy/Local Network with all the other applications that requested local network access.

**Smartphone (please complete the following information):**
 - iPad Pro
 - iOS: 16.2

Clon1998, 2023-01-07:
I will look into that once I have access to an IPad of a friend.
However, I know several people who are running mobileraker on IPad.

chriswhite72, 2023-01-07:
After reading your comment about other people using iPad, I wondered if this is possibly an issue with iPadOS 16.2 or some sort of bad state on my iPad.  So I installed the latest iPadOS **16.3 beta** on my iPad and went through the same steps to add my printer.  To my surprise, I was prompted to allow local network access and mobileraker worked properly on this beta.  I also confirmed that mobileraker was listed under settings/privacy/local network with all the other apps that have requested local network access.  Hence this issue was not on your side but was either some sort of bad state on my iPad or a bug in iPadOS 16.2.

Clon1998, 2023-01-07:
> 

Alright! And excellent that it is working now for you.

closed 2023-01-07 by Clon1998