Clon1998/mobileraker #96: Mobile raker asks for API key with trusted clients - but IP is trusted and accessible

tastyratz, 2023-01-07:
**Describe the bug**
Mobileraker will not allow me to add my printer and connect but shows the error: 
"It seems like you configured trusted clients for moonraker. Please add the API key to the printers settings!"
This seems odd especially since Mainsail won't even let you add an API key.

Moonraker trusted clients list includes the full correct CIDR of the wireless as well as the local DHCP DNS domain under CORS.

The fqdn address is reachable from firefox on this same phone as well as my desktop and it loads successfully, just not the app.

This is MainsailOS with Mainsail and Fluidd installed. I tried different ports for each, connecting via ip vs name, trying the Fluidd port and API key.
Mainsail access log does see the connection and only enters this:
192.168.1.25 - - [06/Jan/2023:23:53:03 -0500] "GET / HTTP/1.1" 200 714 "-" "Dart/2.18 (dart:io)"

**Smartphone (please complete the following information):**
 - Device: Pixel 6
 - OS: Android 11
 - Browser: Firefox

**Mobileraker-Version
2.0.2-76

Clon1998, 2023-01-07:
Can you send me a screenshot of the printer's edit page so I can verify that you configured the websocket addresses correctly?

tastyratz, 2023-01-07:
Here you go:
[image]
And here is a screenshot of the same URL loading successfully on the same device
[image]

Clon1998, 2023-01-07:
Could you add the WebSocket instance endpoint to the WebSocket-address: `ws://mainsailos.local.tastyratz.com` -> `ws://mainsailos.local.tastyratz.com/websocket` and try it again?

Clon1998, 2023-01-07:
> 

I noticed that there might be a bug after changing the ws-url. Can retry it and force close to app and reopen it?

tastyratz, 2023-01-07:
That seems to be the trick! I left the /websocket on the end after your suggestion and restarted the app. I'm now up and running!
Thanks for your help.

closed 2023-01-07 by tastyratz