Clon1998/mobileraker #97: JSON Parse error for printers with multiple extruders

rosscon, 2023-01-07:
**Describe the bug**
Receiving the below error on printers with multiple extruders with shared heaters, suspect it may be due to some of the fields in [config_extruder.dart](https://github.com/Clon1998/mobileraker/blob/master/lib/data/dto/config/config_extruder.dart) are not defined for multiple extruders e.g. nozzle & filament width.

**Screenshots**
App
[image]

Klipper Extruder config
[image]

**Smartphone (please complete the following information):**
iOS 16.1.1

**Mobileraker-Version
2.0.2

Clon1998, 2023-01-07:
I will look into that. Most likely, it is caused by the shared config.
Since I am not running that setup, could you provide me with the JSON provided by the HTTP call `GET http://<YOUR PRINTER IP>/printer/objects/query?configfile` that fetches the entire config and that is actually parsed?

rosscon, 2023-01-07:
awesome cheers :)

json file attached
[configfile.json.zip](https://github.com/Clon1998/mobileraker/files/10366758/configfile.json.zip)

mlee12382, 2023-02-26:
Any update on this issue? I also have dual extruders with a shared nozzle and I am getting the same error.

Clon1998, 2023-02-26:
Sorry,
I was quite busy and the issue got lost.
I just had a look at it and found an error caused by my marshaling layer.
It interpreted the `[extruder_stepper ..]` config as an extruder and therefore, caused the bug!
However, I just committed a fix for that (efc65f0fd0093ae69753abb54b7729ebcaf2f452) and triggered a new release that should resolve that issue. Please let me know if It works now.

And again, apologies for my late response/fix.

mlee12382, 2023-02-26:
@Clon1998 awesome thanks for the update,  I will try it out when it's live and let you know if there are still any issues.

Clon1998, 2023-02-26:
> @Clon1998 awesome thanks for the update,  I will try it out when it's live and let you know if there are still any issues.

iOS version just got approved by apple. Android is still pending

mlee12382, 2023-02-26:
> > @Clon1998 awesome thanks for the update,  I will try it out when it's live and let you know if there are still any issues.
> 
> iOS version just got approved by apple. Android is still pending

Android update just went live and it seems to be working! Thanks again for the help.

closed 2023-03-21 by Clon1998