Clon1998/mobileraker #99: Cannot establish remote connection when off LAN

Belariean, 2023-01-11:
**Describe the bug**
Having issues trying to use Public IP and Port forwarding through Router and Application to gain remote access when using Mobile
attempts at config the mobile companion through SSH have been difficult to follow and complete, details are lacking to fully understand entire process

Belariean, 2023-01-13:
I already have a VPN server running, but any attempts at trying to connect YOUR app directly to Tunnel into my network with out having to first turn on my VPN client on my mobile device then navigate to your app.

so yes, ive attempted Port Forwarding... no luck
attempted putting a OpenVpn Client onto the pi, no luck

all I want to be able to do is while im away from my house, open your app, check on my printer and then make changes (cancel etc) if needed and then continue on my day. What is the point of having a Mobile app that offers connection and control, but only on your own network/ having to rely on a 3rd party VPN client to then use the home network.

Clon1998, 2023-01-13:
Well it's not an advertised feature of the app. The app is open source and free, so if you feel like you need remote access directly in the app, feel free to implement it.
You always have to connect via your vpn client before you're able to connect to your printer via that app.
However, the app is able to receive print status notifications  even if you're not connected to the same network or vpn as your printer.

closed 2023-02-13 by Clon1998