Clon1998/mobileraker #100: Marker for printed objects | Markierung für gedruckte Objekte

Experte9, 2023-01-12:
**Is your feature request related to a problem? Please describe.**
In the app I have no way to see which object I have already printed. I quickly lose track because I always upload multiple files.

In der App habe ich keine Möglichkeit zu schauen welches Objekt ich schon gedruckt habe. Ich verliere schnell die Übersicht da ich immer mehrere Dateien hochlade.

**Describe the solution you'd like**
It would be nice if there was a green check mark, the font turns green or the number of times it was printed somewhere.

Es wäre schön wenn ein grüner Haken, die Schrift grün wird oder die Anzahl wie oft es gedruckt wurde irgendwo steht.

**Describe alternatives you've considered**
Currently I always look at the Moonraker web interface to find out.

Aktuell schaue ich immer auf die Weboberfläche von Moonraker um das in Erfahrung zu bekommen.

**Additional context**
Here are a few pictures of what I mean (Moonraker and Octoprint)
Hier mal ein paar Bilder was ich meine (Moonraker und Octoprint)
[image]
[image]

Clon1998, 2023-01-12:
Hey,
that's a great idea to add sth like that directly to the file browser. Currently, you could be able to look this up by checking if a selected file shows a date or no data for the "last printed" ent

[image]

AustinSchulz, 2023-01-18:
I would suggest adding the indicator as a [badge](https://m3.material.io/components/badges/overview) on top of the print preview image in the list. You can display the number of times completed, and the color be the last print's final state.

Experte9, 2023-02-19:
[image]

closed 2023-02-19 by Experte9