Clon1998/mobileraker #106: Fetching Printer... No Connection

St1fT, 2023-01-25:
Hi,
Mobileraker connect to my Printer, but the Dashboard shows "fetching printer..."
My Config part is: */printer_data/config/
I don´t know, if this is the issue.

akno2205, 2023-02-01:
Hy, same Problem here. After a new installation of Klipper to my printer, with the new config path (printer_data/config. I’m not able to connect with the app to the printer. I get the same error and the following details:

akno2205, 2023-02-01:
Exception:
 type 'Null' is not a subtype of type 'double'

#0      new ConfigExtruder.parse (package:mobileraker/data/dto/config/config_extruder.dart:12)
#1      new ConfigFile.parse (package:mobileraker/data/dto/config/config_file.dart:42)
#2      PrinterService._updateConfigFile (package:mobileraker/service/moonraker/printer_service.dart:757)
#3      PrinterService._parseObjectType (package:mobileraker/service/moonraker/printer_service.dart:426)
#4      PrinterService._temperatureStore.<anonymous closure> (package:mobileraker/service/moonraker/printer_service.dart:377)
#5      _LinkedHashMapMixin.forEach (dart:collection-patch/compact_hash.dart:617)
#6      PrinterService._parseQueriedObjects (package:mobileraker/service/moonraker/printer_service.dart:496)
#7      PrinterService._printerObjectsQuery (package:mobileraker/service/moonraker/printer_service.dart:1032)
<asynchronous suspension>
#8      PrinterService.refreshPrinter (package:mobileraker/service/moonraker/printer_service.dart:168)
<asynchronous suspension>

Failed-Key: configfile 
Raw Json:
{"configfile":{"config":{"respond":{},"delayed_gcode ONLINE_MSG":{"initial_duration":"1.","gcode":"\nRESPOND PREFIX=telegram: MSG=\"Hello. I am online and ready to print :-)\""},"gcode_macro NOZZEL_CLEAN":{"gcode":"\nG90\nG1 Z3.0 X192 F2000\nG1 Z3.0 X180 F2000\nG1 Z3.0 X192 F2000\nG1 Z3.0 X180 F2000\nG1 Z3.0 X192 F2000\nG1 Z3.0 X180 F2000\nG1 Z3.0 X192 F2000\nG1 Z3.0 X180 F2000"},"gcode_macro START_PRINT":{"gcode":"\n{% set T_BED = params.BED_TEMP|default(0)|float %}\n{% set T_EXTRUDER = params.EXTRUDER_TEMP|default(0)|float %}\n\nM140 S{T_BED}\n\nM190 S{T_BED}\n\nM109 S{T_EXTRUDER}\n\nG90\n\nSET_GCODE_OFFSET Z=0.0\n\nG28\n\nBED_MESH_CLEAR\nBED_MESH_CALIBRATE\n\n\n\nG92 E0\nG1 Z5.0 F3000\nG1 X0.1 Y20 Z0.2 F5000.0\nG1 X0.1 Y170.0 Z0.2 F1500.0 E15\nG1 X0.4 Y170.0 Z0.2 F5000.0\nG1 X0.4 Y20 Z0.2 F1500.0 E30\nG92 E0\nG1 Z2.0 F3000\nG1 X5 Y20 Z0.2 F5000.0\nUPDATE_DELAYED_GCODE ID=CASELIGHT_OFF_PRINT DURATION=60"},"gcode_macro END_PRINT":{"gcode":"\n\n\nG91\nG1 E-2 F2700\nG1 E-2 Z0.2 F2400\nG1 X5 Y5 F3000\nG1 Z10\nG90\nG1 X0 Y180 F1500\n\nM140 S0\nM104 S0\nM106 S0\n\nM84\nUPDATE_DELAYED_GCODE ID=CASELIGHT_ON_PRINT DURATION=1"},"gcode_macro G29":{"gcode":"\nM190 S60\nG28\nBED_MESH_CLEAR\nBED_MESH_CALIBRATE\nG0 X0 Y0 Z10 F6000"},"gcode_macro LOAD_FILAMENT":{"gcode":"\n\nG91\nG92 E0\nG1 E300 F900\nG1 E75 F100"},"gcode_macro UNLOAD_FILAMENT":{"gcode":"\n\nG91\nG92 E0\nG1 E25 F100\nG1 E-375 F900"},"gcode_macro SCREW1":{"gcode":"\nG90\nG1 X25 Y70 Z30 F1500"},"gcode_macro SCREW2":{"gcode":"\nG90\nG1 X155 Y70 Z30 F1500"},"gcode_macro SCREW3":{"gcode":"\nG90\nG1 X155 Y185 Z30 F1500"},"gcode_macro SCREW4":{"gcode":"\nG90\nG1 X25 Y185 Z30 F1500"},"virtual_sdcard":{"path":"~/printer_data/gcodes","on_error_gcode":"CANCEL_PRINT"},"pause_resume":{},"display_status":{},"gcode_macro CANCEL_PRINT":{"description":"Cancel the actual running print","rename_existing":"CANCEL_PRINT_BASE","gcode":"\n\n{% set macro_found = True if printer['gcode_macro _CLIENT_VARIABLE'] is defined else False %}\n{% set client = printer['gcode_macro _CLIENT_VARIABLE'] %}\n{% set allow_park = False if not macro_found\nelse False if client.park_at_cancel is not defined\nelse True  if client.park_at_cancel|lower == 'true'\nelse False %}\n{% set retract      = 5.0  if not macro_found else client.cancel_retract|default(5.0)|abs %}\n{% set sp_retract   = 2100 if not macro_found else client.speed_retract|default(35) * 60 %}\n\n{% if not printer.pause_resume.is_paused and allow_park %} _TOOLHEAD_PARK_PAUSE_CANCEL {% endif %}\n{% if printer.extruder.can_extrude %}\nM83\nG1 E-{retract} F{sp_retract}\n{% if printer.gcode_move.absolute_extrude %} M82 {% endif %}\n{% else %}\n{action_respond_info(\"Extruder not hot enough\")}\n{% endif %}\nTURN_OFF_HEATERS\nM106 S0\nCANCEL_PRINT_BASE"},"gcode_macro PAUSE":{"description":"Pause the actual running print","rename_existing":"PAUSE_BASE","gcode":"\nPAUSE_BASE\n_TOOLHEAD_PARK_PAUSE_CANCEL {rawparams}"},"gcode_macro RESUME":{"description":"Resume the actual running print","rename_existing":"RESUME_BASE","gcode":"\n\n{% set macro_found = True if printer['gcode_macro _CLIENT_VARIABLE'] is defined else False %}\n{% set client = printer['gcode_macro _CLIENT_VARIABLE'] %}\n{% set velocity = printer.configfile.settings.pause_resume.recover_velocity %}\n{% set use_fw_retract = False if not macro_found\nelse False if client.use_fw_retract is not defined\nelse True  if client.use_fw_retract|lower == 'true' and printer.firmware_retraction is defined\nelse False %}\n{% set unretract      = 1.0  if not macro_found else client.unretract|default(1.0)|abs %}\n{% set sp_unretract   = 2100 if not macro_found else client.speed_unretract|default(35) * 60 %}\n{% set sp_move        = velocity if not macro_found else client.speed_move|default(velocity) %}\n\n{% if printer.extruder.can_extrude %}\n{% if use_fw_retract %}\nG11\n{% else %}\nM83\nG1 E{unretract} F{sp_unretract}\n{% if printer.gcode_move.absolute_extrude %} M82 {% endif %}\n{% endif %}\n{% else %}\n{action_respond_info(\"Extruder not hot enough %s\" % use_fw_retract)}\n{% endif %}\nRESUME_BASE VELOCITY={params.VELOCITY|default(sp_move)}"},"gcode_macro _TOOLHEAD_PARK_PAUSE_CANCEL":{"description":"Helper: park toolhead used in PAUSE and CANCEL_PRINT","gcode":"\n\n{% set macro_found = True if printer['gcode_macro _CLIENT_VARIABLE'] is defined else False %}\n{% set client = printer['gcode_macro _CLIENT_VARIABLE'] %}\n{% set velocit [... cut ...] lhead":{"homed_axes":"xyz","axis_minimum":[-10.0,0.0,-4.0,0.0],"axis_maximum":[183.0,186.0,200.0,0.0],"print_time":9824.369963536523,"stalls":0,"estimated_print_time":9822.313905733334,"extruder":"extruder","position":[72.511,101.361,0.7458177747706348,5060.06363],"max_velocity":300.0,"max_accel":3000.0,"max_accel_to_decel":3000.0,"square_corner_velocity":5.0},"extruder":{"temperature":214.95,"target":215.0,"power":0.7684993808744486,"can_extrude":true,"pressure_advance":0.0,"smooth_time":0.04}}

Clon1998, 2023-02-01:
I will have a look at that once my only Klipper printer is back up running. Otherwise it's hard for me to reproduce it.

akno2205, 2023-02-01:
Thank you for the fast reply.

klHaribo, 2023-02-24:
I have the same error.
printer worked fine with mobileraker. 
 until I changed my setup to dual color, this error came up. 
I am now on the last version from github

Clon1998, 2023-02-24:
I currently don't have time to look into that since I have a few finals coming up the next week's.
I am pretty sure there is a issue with the config parsing?

klHaribo, 2023-02-25:
I am not a programmer, but I can take a look at it in the next days. I am also very busy :)

Clon1998, 2023-02-26:
Just noticed that this issue is related to #97, or at least the problem @akno2205, @klHaribo encountered.

I just fixed that issue, as mentioned in #97 a new app version `2.1.1` should be available in the next few hours.

Clon1998, 2023-02-26:
However, this does not resolve the issue the original author @St1fT has AFAIK.

Therefore, I will look more into that once I am done with my exams in the upcoming weeks.
If I forget it, don't hesitate to ping me again @St1fT !

freeblazin, 2023-03-14:
I am also experiencing the OP's symptom.  Dashboard shows "fetching printer..." indefinitely.  Console commands and emergency stop from app are working as expected, confirming connection to moonraker api is good.  I am also running the latest moonraker version with changes in config directory structure to "`/printer_data/config/".  I tested mobileraker v2.1.1 with no change in symptom.

Hope your exams went well good sir!!  Let me know if I can support any other testing or data/log gathering to support when you get around to looking into this.

Clon1998, 2023-03-14:
> I am also experiencing the OP's symptom.  Dashboard shows "fetching printer..." indefinitely.  Console commands and emergency stop from app are working as expected, confirming connection to [...]

Are you by any chance using iOS?
If that's the case can you try the latest beta:

https://testflight.apple.com/join/h9JwuyR0

And if it is not working, can you send me the log files that are accessible at the bottom of the Settings page?

freeblazin, 2023-03-14:
I am using Android app version.  Looks like there are no log files in the settings page for the Android app. Is there another way to access the app logs?

Clon1998, 2023-03-15:
> I am using Android app version. Looks like there are no log files in the settings page for the Android app. Is there another way to access the app logs?

Can you try to join the beta program on PlayStore? 
I just double-checked it and by now, google should have approved the latest build (2.1.2-126) for the public.

ripinteer, 2023-03-25:
> https://testflight.apple.com/join/h9JwuyR0 <- I tried this version and stable from App Store (at 25/03/2023) still "fetching printer"

I also have indefinite "fetching printer" status. The connection seems to be established and the console is working properly. I have collected the contents of the "debug info" section, and it seems like there is something wrong with `No remote client configured... cant do handover!`
IOS 14.8.1
[deubg.txt](https://github.com/Clon1998/mobileraker/files/11068530/deubg.txt)
[image]

Clon1998, 2023-03-25:
> I also have indefinite "fetching printer" status. The connection seems to be established and the console is working properly. I have collected the contents of the "debug info" section, and it seems [...]

Hey thanks for the log, I will look into that.
Regarding the message `[ [38;5;12m💡 No remote client configured... can't do handover! ref:92050206 [0m]`, this message simply means that no OE client is available for usage in case the current local client is unable to connect to your machine. However, as the log entry `[ [38;5;12m💡 [516785204-ws://192.168.2.119/websocket] ClientState.connecting ➝ ClientState.connected [0m]` shows, your local client is working fine and therefore this has no influence.

Clon1998, 2023-03-25:
> I also have indefinite "fetching printer" status. The connection seems to be established and the console is working properly. I have collected the contents of the "debug info" section, and it seems [...]

I had a brief look at the output, it seems like the app gets stuck during the parsing of the cached temperature data.
Can you do me favor and send me the output of the following web-request in order for me to debug/reproduce it locally:

1. `http://192.168.2.119/printer/objects/list`
2. `http://192.168.2.119/server/temperature_store

ripinteer, 2023-03-25:
Thanks for explanation. One more thing, I remember that I faced this problem last year. Unfortunately, I was busy studying, so I didn't report the problem then.

---
As for the jsons, here they are:

[temperature_store.txt](https://github.com/Clon1998/mobileraker/files/11068605/temperature_store.txt)
[list.txt](https://github.com/Clon1998/mobileraker/files/11068606/list.txt)

I changed .json to .txt, cause github don't like it for some reason.

Clon1998, 2023-03-25:
> Thanks for explanation. One more thing, I remember that I faced this problem last year. Unfortunately, I was busy studying, so I didn't report the problem then.
>
> As for the jsons, here they are: [...]

Thanks for that!
I will plug the json directly into some of my code and see what happens.

Clon1998, 2023-03-25:
> Thanks for explanation. One more thing, I remember that I faced this problem last year. Unfortunately, I was busy studying, so I didn't report the problem then.
>
> As for the jsons, here they are: [...]
I think I already found the issue!
Is there any reason to why you did not configure any `heater_bed`?

ripinteer, 2023-03-25:
>I think I already found the issue!
>Is there any reason to why you did not configure any heater_bed?

That's right, I have my test bench on my table, and I removed `heater_bed` from config, so Klipper won't complain :)

Clon1998, 2023-03-25:
> > I think I already found the issue!
> > Is there any reason to why you did not configure any heater_bed?
> 
> That's right, I have my test bench on my table, and I removed `heater_bed` from config, so Klipper won't complain :)

Alright, I will hopefully have some time today to remove all of my hardcoded `heater_bed` references that are causing/will cause errors!

ripinteer, 2023-03-25:
Hooked up the bed and config piece)
[image]

Clon1998, 2023-03-25:
Fixed in [2e598ee](https://github.com/Clon1998/mobileraker/commit/2e598ee009ff8995f1673465c49a8ceb26a11300)!
Should be available in the stores in the next 24-72hrs.
For Android users, feel free to sideload the app via [GitHub](https://github.com/Clon1998/mobileraker/releases/tag/android2.2.1) !

closed 2023-03-25 by Clon1998