Clon1998/mobileraker #110: No enclosure temperature reading if 'temperature_fan' is used instead of 'temperature_sensor'

nichevy, 2023-01-29:
I get temperature readings from all sensors that are set as temperature_sensor but nothing from temperature_fan that is used to control the exhaust fan based on temperature readings as described below:

[temperature_fan chamber]
pin: PD13
max_power: 0.6
shutdown_speed: 0.0
kick_start_time: 5.0
cycle_time:0.01
off_below:0.1
sensor_type: Generic 3950
sensor_pin: PF5
min_temp: 0
max_temp: 70
target_temp: 45.0
control: watermark
gcode_id: C

**Smartphone (please complete the following information):**
 - Device: Huawei CLT-L29
 - OS: Android 10

**Mobileraker-Version
2.1.0.

Clon1998, 2023-01-29:
Related to #98

Clon1998, 2023-02-13:
Close b3221adb7d54cc7423905e791c2b3ebd3697486d

closed 2023-02-13 by Clon1998