Clon1998/mobileraker #115: Printer Tabs

Avatarsia, 2023-02-10:
I have a Design Request. 
For Wide Displays like Tablets it would be Amazing to have Tabs to Change Printers at the Top
Printer A Dashboard, Printer B Dashboard etc.

Clon1998, 2023-02-10:
Hey,
that's quite a good idea!
However, since I don't own any tablet device this is quite low priority.
However, there is a shortcut you might like.
In the dashboard, console and files view you are able to swipe the title/header to switch between multiple printers 😉

Avatarsia, 2023-02-10:
> Hey, that's quite a good idea! However, since I don't own any tablet device this is quite low priority. However, there is a shortcut you might like. In the dashboard, console and files view you are able to swipe the title/header to switch between multiple printers 😉

when i open the side menu, i cant swipe anything, may it be that this works just with newer android versions ?

Clon1998, 2023-02-10:
Try to swipe here

[image]

Avatarsia, 2023-02-10:
nope dont liks it, but i have also to check for the moment why my dashboard is non stop loading. 
maybe cuz of this it cant change the dashboards

Clon1998, 2023-02-10:
> nope dont liks it, but i have also to check for the moment why my dashboard is non stop loading. 
> 
> maybe cuz of this it cant change the dashboards

Interesting...
I am currently not in reach of my laptop where I have a better test setup.
I will verify that it the swipe still works once i am back home.

Avatarsia, 2023-02-10:
> Interesting... I am currently not in reach of my laptop where I have a better test setup. I will verify that it the swipe still works once i am back home.

my bad, i just tried it on the phone and realized you mean the tab i wanted to be duplicated :D
it works on phone and tablet

Avatarsia, 2023-02-10:
but maybe you could check whats gone wrong here, actually the star link gcode isnt even from this printer

2023-02-10 14:11:02,743 [app.py:log_request()] - 404 GET /machine/device_power/devices (127.0.0.1) [No User] 4.40ms
2023-02-10 14:11:03,157 [file_manager.py:get_file_list()] - Updating File List <gcodes>...
2023-02-10 14:11:04,306 [klippy_connection.py:_request_initial_subscriptions()] - Webhooks Subscribed
2023-02-10 14:11:04,311 [klippy_connection.py:_request_initial_subscriptions()] - GCode Output Subscribed
2023-02-10 14:11:04,315 [app.py:register_remote_handler()] - Registering HTTP endpoint: (GET POST) /printer/motion_report/dump_trapq
2023-02-10 14:11:04,316 [websockets.py:register_api_handler()] - Registering Websocket JSON-RPC methods: printer.motion_report.dump_trapq
2023-02-10 14:11:04,553 [job_state.py:_handle_started()] - Job state initialized: standby
2023-02-10 14:11:04,827 [klippy_connection.py:_check_ready()] - Klippy ready
2023-02-10 14:11:05,306 [data_store.py:_init_sensors()] - Configuring available sensors: ['extruder']
2023-02-10 14:14:28,877 [websockets.py:on_close()] - Websocket Closed: ID: 1749977328 Close Code: None, Close Reason: None, Pong Time Elapsed: 0.29
2023-02-10 14:14:30,127 [app.py:log_request()] - 101 GET /websocket (192.168.0.49) [_TRUSTED_USER_] 6.07ms
2023-02-10 14:14:30,128 [websockets.py:open()] - Websocket Opened: ID: 1729462224, Proxied: True, User Agent: Dart/2.18 (dart:io), Host Name: 192.168.0.26
2023-02-10 14:14:36,048 [websockets.py:build_error()] - JSON-RPC Request Error: -32601
Key 'settings' in namespace 'mobileraker' not found
Traceback (most recent call last):
  File "/home/pi/moonraker/moonraker/websockets.py", line 300, in execute_method
    result = await callback(params)
  File "/home/pi/moonraker/moonraker/websockets.py", line 416, in func
    ip_addr=sc.ip_addr, user=sc.user_info))
  File "/home/pi/moonraker/moonraker/components/database.py", line 765, in _handle_item_request
    val = await self.get_item(namespace, key)
  File "/usr/lib/python3.7/concurrent/futures/thread.py", line 57, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/home/pi/moonraker/moonraker/components/database.py", line 199, in func_wrapper
    return command_func(*args)
  File "/home/pi/moonraker/moonraker/components/database.py", line 362, in _get_impl
    ns = self._get_record(namespace, key_list[0])
  File "/home/pi/moonraker/moonraker/components/database.py", line 678, in _get_record
    f"Key '{key}' in namespace '{namespace}' not found", 404)
utils.ServerError: Key 'settings' in namespace 'mobileraker' not found
2023-02-10 14:27:06,710 [websockets.py:build_error()] - JSON-RPC Request Error: -32601
Metadata not available for <Left_Front_Starlink_G3_UT_Flat_Mount_ASA_10h47m.gcode>
Traceback (most recent call last):
  File "/home/pi/moonraker/moonraker/websockets.py", line 300, in execute_method
    result = await callback(params)
  File "/home/pi/moonraker/moonraker/websockets.py", line 416, in func
    ip_addr=sc.ip_addr, user=sc.user_info))
  File "/home/pi/moonraker/moonraker/components/file_manager/file_manager.py", line 366, in _handle_metadata_request
    f"Metadata not available for <{requested_file}>", 404)
utils.ServerError: Metadata not available for <Left_Front_Starlink_G3_UT_Flat_Mount_ASA_10h47m.gcode>
2023-02-10 14:27:06,722 [websockets.py:build_error()] - JSON-RPC Request Error: -32601
Key 'settings' in namespace 'mobileraker' not found
Traceback (most recent call last):
  File "/home/pi/moonraker/moonraker/websockets.py", line 300, in execute_method
    result = await callback(params)
  File "/home/pi/moonraker/moonraker/websockets.py", line 416, in func
    ip_addr=sc.ip_addr, user=sc.user_info))
  File "/home/pi/moonraker/moonraker/components/database.py", line 765, in _handle_item_request
    val = await self.get_item(namespace, key)
  File "/usr/lib/python3.7/concurrent/futures/thread.py", line 57, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/home/pi/moonraker/moonraker/components/database.py", line 199, in func_wrapper
    return command_func(*args)
  File "/home/pi/moonraker/moonraker/components/database.py", line 362, in _get_impl
    ns = self._get_record(namespace, key_list[0])
  File "/home/pi/moonraker/moonraker/components/database.py", line 678, in _get_record
    f"Key '{key}' in namespace '{namespace}' not found", 404)
utils.ServerError: Key 'settings' in namespace 'mobileraker' not found
2023-02-10 14:28:03,275 [websockets.py:build_error()] - JSON-RPC Request Error: -32601
Metadata not available for <Left_Front_Starlink_G3_UT_Flat_Mount_ASA_10h47m.gcode>
Traceback (most recent call last):
  File "/home/pi/moonraker/moonraker/websockets.py", line 300, in execute_method
    result = await callback(params)
  File "/home/pi/moonraker/moonraker/websockets.py", line 416, in func
    ip_addr=sc.ip_addr, user=sc.user_info))
  File "/home/pi/moonraker/moonraker/components/file_manager/file_manager.py", line 366, in _handle_metadata_request
    f"Metadata not available for <{requested_file}>", 404)
utils.ServerError: Metadata not available for <Left_Front_Starlink_G3_UT_Flat_Mount_ASA_10h47m.gcode>
2023-02-10 14:28:03,283 [websockets.py:build_error()] - JSON-RPC Request Error: -32601
Key 'settings' in namespace 'mobileraker' not found
Traceback (most recent call last):
  File "/home/pi/moonraker/moonraker/websockets.py", line 300, in execute_method
    result = await callback(params)
[... cut ...]
2023-02-10 14:28:39,076 [websockets.py:on_close()] - Websocket Closed: ID: 1750028208 Close Code: 1001, Close Reason: None, Pong Time Elapsed: 0.21
2023-02-10 14:28:39,486 [app.py:log_request()] - 101 GET /websocket (192.168.0.20) [_TRUSTED_USER_] 7.13ms
2023-02-10 14:28:39,487 [websockets.py:open()] - Websocket Opened: ID: 1729522896, Proxied: True, User Agent: Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/110.0.0.0 Safari/537.36, Host Name: 192.168.0.26
2023-02-10 14:28:39,495 [websockets.py:_handle_identify()] - Websocket 1729522896 Client Identified - Name: mainsail, Version: 2.4.1, Type: web

Clon1998, 2023-02-10:
That log/error is from moonraker or from my app?

Avatarsia, 2023-02-10:
Moonraker, your app have a log to access ?

Clon1998, 2023-02-10:
No log access.
But in case of an error you will get quite a large stack trace

Avatarsia, 2023-02-10:
> No log access. But in case of an error you will get quite a large stack trace

i can access logs, gcode, only dashboard is non stop loading

Clon1998, 2023-02-10:
I will look at the moonraker log to verify that.

By any means, are you using a single Pi to Host two printers?

Avatarsia, 2023-02-10:
> I will look at the moonraker log to verify that.
> 
> By any means, are you using a single Pi to Host two printers?

i have also that, but in this case we have 2 different mashine's

Clon1998, 2023-02-10:
I will have a look at the moonraker log once i am back home.
I think you might have a faulty http url configured inside the app.

Avatarsia, 2023-02-10:
> I will have a look at the moonraker log once i am back home. I think you might have a faulty http url configured inside the app.

i just enteres the ip

Avatarsia, 2023-02-13:
> I will have a look at the moonraker log once i am back home. I think you might have a faulty http url configured inside the app.

did you had time to check it out ?

Clon1998, 2023-02-13:
Hey,
I just looked at it.
Did you use an older version of mobileraker before?

Clon1998, 2023-02-13:
I dug a little bit deeper.
So, there is a problem with the settings my app stores in the database of moonraker.
I remember a bug that caused the app to go into an infinite loop in case the app was a new setup. However, I just tested it with my current dev version and the latest version that should be available through the stores and both of them work.
Therefore, could you provide me with the version number of the app version you are currently running?

Avatarsia, 2023-02-13:
moonraker
v0.7.1-868-g0de712e8
mobileraker 
2.1.0-97

Avatarsia, 2023-03-21:
> I dug a little bit deeper. So, there is a problem with the settings my app stores in the database of moonraker. I remember a bug that caused the app to go into an infinite loop in case the app was a [...]

did you had any chance to go after it, i just made a fw update of the pi but with this printer the issue still remain

Clon1998, 2023-03-21:
Sorry,
i got distracted by exams and other stuff like octoeverywhere support.

Just to catch up again. You are experiencing the issue only on one of your printers?

Can you provide me the actual screen that is shown on the Dashboard?

In addition please provide me with the content of the following requests if you enter them into your browser:

1. `http://<YOUR IP>/server/info`
2. `http://<YOUR IP>/printer/info`
3. `http://<YOUR IP>/printer/objects/list`
4. `http://<YOUR IP>/printer/objects/query?configfile`

Avatarsia, 2023-03-23:
/server/info
{"result": {"klippy_connected": true, "klippy_state": "ready", "components": ["klippy_connection", "application", "websockets", "internal_transport", "dbus_manager", "database", "file_manager", "klippy_apis", "secrets", "template", "shell_command", "machine", "data_store", "proc_stats", "job_state", "job_queue", "http_client", "announcements", "webcam", "extensions", "update_manager", "authorization", "octoprint_compat"], "failed_components": [], "registered_directories": ["config", "logs", "gcodes", "config_examples", "docs"], "warnings": [], "websocket_count": 5, "moonraker_version": "v0.8.0-29-g80920dd", "missing_klippy_requirements": [], "api_version": [1, 2, 1], "api_version_string": "1.2.1"}}

printer/info
{"result": {"state_message": "Printer is ready", "klipper_path": "/home/pi/klipper", "config_file": "/home/pi/klipper_config/printer.cfg", "software_version": "v0.11.0-128-g57c4da5e", "hostname": "toasty", "cpu_info": "4 core ARMv7 Processor rev 4 (v7l)", "state": "ready", "python_path": "/home/pi/klippy-env/bin/python", "log_file": "/home/pi/klipper_logs/klippy.log"}}

/printer/objects/list
{"result": {"objects": ["webhooks", "configfile", "mcu", "mcu zboard", "pause_resume", "display_status", "gcode_macro CANCEL_PRINT", "gcode_macro PAUSE", "gcode_macro RESUME", "gcode_macro BED_MESH_CALIBRATE", "fan", "heaters", "heater_fan my_nozzle_fan", "gcode_move", "probe", "bed_mesh", "print_stats", "virtual_sdcard", "idle_timeout", "gcode_macro START_PRINT", "gcode_macro END_PRINT", "gcode_macro BED_TRIMMING", "quad_gantry_level", "gcode_macro ECKEN_LEVELN_MANUELL", "gcode_macro ECKEN_LEVELN_PROBE", "gcode_macro Z_OFFSET", "gcode_macro BLTOUCH_CALIBRATE", "gcode_macro UNLOAD_FILAMENT", "gcode_macro LOAD_FILAMENT", "gcode_macro G29", "stepper_enable", "motion_report", "query_endstops", "system_stats", "manual_probe", "toolhead", "extruder"]}}

/printer/objects/query?configfile
{"result": {"status": {"configfile": {"save_config_pending_items": {}, "warnings": [], "config": {"gcode_macro BED_MESH_CALIBRATE": {"rename_existing": "_BED_MESH_CALIBRATE", "variable_last_area_end_x": "-1", "variable_last_area_end_y": "-1", "gcode": "\n{% if params.FORCE_NEW_MESH != null %}\n{ action_respond_info(\"Force New Mesh: %s\" % (params.FORCE_NEW_MESH)) }\n{% endif %}\n{% if printer[\"bed_mesh\"].profile_name == '' %}\n{ action_respond_info(\"No existing bed mesh found.\") }\n{% set last_area_end_x=-1 %}\n{% endif %}\n{% if printer.toolhead.homed_axes != \"xyz\" %}\nG28\n{% endif %}\n{% set klicky_available = printer['gcode_macro _Probe_Variables'] != null %}\n{% set euclid_available = printer['gcode_macro EuclidProbe'] != null %}; Requires v5 macros https://github.com/nionio6915/Euclid_Probe/blob/main/Firmware_Examples/Klipper/00-euclid_exampleV5.cfg\n{% if params.PRINT_MIN %}\n{ action_respond_info(\"print_min: %s\" % params.PRINT_MIN) }\n{ action_respond_info(\"print_max: %s\" % params.PRINT_MAX) }\n{% set blTouchConfig = printer['configfile'].config[\"bltouch\"] %}\n{% if blTouchConfig %}\n{% set OffsetX = blTouchConfig.x_offset|default(0)|float %}\n{% set OffsetY = blTouchConfig.y_offset|default(0)|float %}\n{% endif %}\n{% set probeConfig = printer['configfile'].config[\"probe\"] %}\n{% if probeConfig %}\n{% set OffsetX = probeConfig.x_offset|default(0)|float %}\n{% set OffsetY = probeConfig.y_offset|default(0)|float %}\n{% endif %}\n{% set print_min_x = params.PRINT_MIN.split(\",\")[0]|float %}\n{% set print_min_y = params.PRINT_MIN.split(\",\")[1]|float %}\n{% set print_max_x = params.PRINT_MAX.split(\",\")[0]|float %}\n{% set print_max_y = params.PRINT_MAX.split(\",\")[1]|float %}\n{% if last_area_start_x > 0 %}\n{ action_respond_info(\"last_bed_mesh: %s,%s %s,%s\" % (last_area_start_x, last_area_start_y, last_area_end_x, last_area_end_y)) }\n{% endif %}\n{% if (params.FORCE_NEW_MESH != null) or (print_min_x < last_area_start_x) or (print_max_x > last_area_end_x) or (print_min_y < last_area_start_y) or (print_max_y > last_area_end_y)  %}\n{% if klicky_available %}\n_CheckProbe action=query\nAttach_Probe\n{% elif euclid_available %}\nDEPLOY_PROBE\n{% endif %}\n{% if (print_min_x < print_max_x) and (print_min_y < print_max_y) %}\n{% set bedMeshConfig = printer['configfile'].config[\"bed_mesh\"] %}\n{% set minimum_probe_count = 3 %}\n{% if bedMeshConfig.algorithm == \"bicubic\" %}\n{% set minimum_probe_count = 5 %}\n{% endif %}\n{% set probe_count = bedMeshConfig.probe_count.split(\",\") %}\n{% set probe_count_x = probe_count[0]|int %}\n{% if probe_count.__len__() > 1 %}\n{% set probe_count_y = probe_count[1]|int %}\n{% else %}\n{% set probe_count_y = probe_count_x|int %}\n{% endif %}\n{% set relative_reference_index = bedMeshConfig.relative_reference_index %}\n{% set mesh_min_x = bedMeshConfig.mesh_min.split(\",\")[0]|float %}\n{% set mesh_min_y = bedMeshConfig.mesh_min.split(\",\")[1]|float %}\n{% set mesh_max_x = bedMeshConfig.mesh_max.split(\",\")[0]|float %}\n{% set mesh_max_y = bedMeshConfig.mesh_max.split(\",\")[1]|float %}\n{% if print_max_x - print_min_x < (mesh_max_x - mesh_min_x) * 0.50 %}\n{% set probe_count_x = minimum_probe_count %}\n{% endif %}\n{% if print_max_y - print_min_y < (mesh_max_y - mesh_min_y) * 0.50 %}\n{% set probe_count_y = minimum_probe_count %}\n{% endif %}\n{% if print_min_x - buffer >= mesh_min_x %}\n{% set mesh_min_x = print_min_x - buffer %}\n{% endif %}\n{% if print_min_y - buffer >= mesh_min_y %}\n{% set mesh_min_y = print_min_y - buffer %}\n{% endif %}\n{% i [... cut ...] ous_corner_velocity": 1.0, "full_steps_per_rotation": 200, "heater_pin": "PH5", "pressure_advance_smooth_time": 0.04, "smooth_time": 7.0, "pid_ki": 12.95, "filament_diameter": 1.75, "sensor_pin": "zboard:PK6", "nozzle_diameter": 1.0, "pid_kd": 61.264, "rotation_distance": 22.6694, "step_pin": "zboard:PF0", "max_power": 1.0, "microsteps": 16, "min_extrude_temp": 180.0, "max_extrude_only_accel": 3326.0135046143027, "max_temp": 330.0}}, "save_config_pending": false}}, "eventtime": 257284.14508274}}

Avatarsia, 2023-03-23:
[image]

Clon1998, 2023-03-23:
> list

Thanks for that...
Sadly all of these files passed my local tests indicating that there is no error in the data layer of the app.

Can you show me a screenshot of the dashboard of the printer that is not working? 

Additionally, could you assign another IP to the machine and try again connecting it to mobileraker?

Clon1998, 2023-03-24:
> to change the ip had no effect i tried that already before
> 
> 
> 
> by the way, the mailfunction messages like "below min temp" works anyway, i also can go into the files or terminal menu
> 
> 
> 
> 
> 
> [image]
> 
> 

Alright...
Hopefully I can look deeper into that today...
At the mean time, did you try the latest beta version available through the Playstore?
It should contain a lot viewer in the app settings that might allow me to take a closer look at what the app is actually doing...

Avatarsia, 2023-03-24:
2.1.2-126 is running atm

Clon1998, 2023-03-25:
> 2.1.2-126 is running atm

Hey one user provided me with the log of his printer that gets stuck!
Therfore, can you also provide me with the output of the temperature_store:

1. `http://<IP>/server/temperature_store`

Clon1998, 2023-03-25:
By any chance, did you configure/not configure a `heater_bed` on that machine?

Avatarsia, 2023-03-25:
> By any chance, did you configure/not configure a `heater_bed` on that machine?

yes there is no heater bed on this mashine

Clon1998, 2023-03-25:
> > By any chance, did you configure/not configure a `heater_bed` on that machine?
> 
> yes there is no heater bed on this mashine

Alright, I will hopefully have some time today to remove all of my hardcoded `heater_bed` references that are causing/will cause errors!

Clon1998, 2023-03-25:
Fixed in [2e598ee](https://github.com/Clon1998/mobileraker/commit/2e598ee009ff8995f1673465c49a8ceb26a11300)!
Should be available in the stores in the next 24-72hrs.
If you are on Andorid, feel free to sideload the app via [GitHub](https://github.com/Clon1998/mobileraker/releases/tag/android2.2.1) as soon as it is ready (about 15 min)!

closed 2023-03-25 by Clon1998

Avatarsia, 2023-03-25:
works just fine, perfect !