Clon1998/mobileraker #117: Webcam not visible/slow on iPadOS

katosabi, 2023-02-14:
**Describe the bug**
First of all, thanks for all the awesome work!

After the latest Mobileraker update, the webcam view on the ipad version of the app is either not visible, or it only loads 1-2 frames and locks up. The same printer and Mobileraker configuration works fine on my iphone. I have also tried this on my wife's ipad and it acts the same; not viewable or locks up. I have tried this with both adaptive stream and stream modes with the same results on both. I have attached screenshots of both the ipad and iphone.

**Smartphone (please complete the following information):**
 - Device: Ipad Pro 4th Gen
 - OS: iPadOS
 - Version 16.3

**Mobileraker-Version**
2.1.0-101

**Screenshots**
[image]

[image]

Clon1998, 2023-02-14:
Hey,
I currently don't own any Ipad version. Still, I think I can look into that using the Simulator provided by Apple. If that does not help, I might ask my friend to lend me his IPad for that purpose.

Anyhow, does the app behave strangely if the Webcam is offline or does everything else just work?

katosabi, 2023-02-14:
It did crash twice on mine and once on her ipad while trying to make the webcam work (I think while saving settings, but not 100% sure), but other than that the app works normally.

Clon1998, 2023-03-21:
Hey @katosabi, I am not sure if you are a member of the IOS beta group but if that's not the case can you please join it and give me feedback if the webcam issue you encountered is resolved?

https://testflight.apple.com/join/h9JwuyR0

katosabi, 2023-03-26:
Hi, sorry for the delay. After switching to Crowsnest (due to upgrading Mainsail) it works as it should now. Closing this. Thanks for looking into it!

closed 2023-03-26 by katosabi