Clon1998/mobileraker #120: Issue opening one printer suddenly.

ChrisWhite1985, 2023-02-28:
**Describe the bug**
When I go to open Voron Printer, I get this error bit not on RatRig RatOS Printer. Sudddenly stopped working.

**Expected behavior**
Open page Mobileraker - worked till recently.

**Mobileraker-Version
Latest.

**Additional context**
Error report from the App.

Exception:
 type '_OneByteString' is not a subtype of type 'List<dynamic>' in type cast

#0      _$$_ConfigNeopixelFromJson (package:mobileraker/data/dto/config/led/config_neopixel.g.dart:20)
#1      new _$_ConfigNeopixel.fromJson (package:mobileraker/data/dto/config/led/config_neopixel.freezed.dart:233)
#2      _$ConfigNeopixelFromJson (package:mobileraker/data/dto/config/led/config_neopixel.freezed.dart:18)
#3      new ConfigNeopixel.fromJson (package:mobileraker/data/dto/config/led/config_neopixel.dart:29)
#4      new ConfigFile.parse (package:mobileraker/data/dto/config/config_file.dart:68)
#5      PrinterService._updateConfigFile (package:mobileraker/service/moonraker/printer_service.dart:842)
#6      PrinterService._parseObjectType (package:mobileraker/service/moonraker/printer_service.dart:465)
#7      PrinterService._temperatureStore.<anonymous closure> (package:mobileraker/service/moonraker/printer_service.dart:415)
#8      _LinkedHashMapMixin.forEach (dart:collection-patch/compact_hash.dart:625)
#9      PrinterService._parseQueriedObjects (package:mobileraker/service/moonraker/printer_service.dart:555)
#10     PrinterService._printerObjectsQuery (package:mobileraker/service/moonraker/printer_service.dart:1139)
<asynchronous suspension>
#11     PrinterService.refreshPrinter (package:mobileraker/service/moonraker/printer_service.dart:197)
<asynchronous suspension>

Failed-Key: configfile 
Raw Json:
{"virtual_sdcard":{"progress":1.0,"file_position":1392584,"is_active":false,"file_path":null,"file_size":1392584},"temperature_sensor mcu_temp":{"measured_min_temp":22.37,"measured_max_temp":27.75,"temperature":22.57},"display_status":{"progress":1.0,"message":"Print Finished"},"configfile":{"warnings":[],"config":{"tmc2209 extruder":{"sense_resistor":"0.110","interpolate":"false","stealthchop_threshold":"0","run_current":"0.5","uart_pin":"PD7","hold_current":"0.2"},"gcode_macro BED_MESH_CALIBRATE":{"rename_existing":"BED_MESH_CALIBRATE_ORIGINAL","gcode":"\nQUERY_PROBE\nM401\nBED_MESH_CALIBRATE_ORIGINAL\nM402"},"gcode_macro CURRENT_NORMAL":{"gcode":"\nM117 Decreased motor currents.\nSET_TMC_CURRENT STEPPER=stepper_x CURRENT={ printer.configfile.config[\"tmc2209 stepper_x\"][\"run_current\"] }\nSET_TMC_CURRENT STEPPER=stepper_y CURRENT={ printer.configfile.config[\"tmc2209 stepper_y\"][\"run_current\"] }\nSET_TMC_CURRENT STEPPER=stepper_z CURRENT={ printer.configfile.config[\"tmc2209 stepper_z\"][\"run_current\"] }\nSET_TMC_CURRENT STEPPER=stepper_z1 CURRENT={ printer.configfile.config[\"tmc2209 stepper_z1\"][\"run_current\"] }\nSET_TMC_CURRENT STEPPER=stepper_z2 CURRENT={ printer.configfile.config[\"tmc2209 stepper_z2\"][\"run_current\"] }\nSET_TMC_CURRENT STEPPER=stepper_z3 CURRENT={ printer.configfile.config[\"tmc2209 stepper_z3\"][\"run_current\"] }","description":"Lowers stepper current for NON-printing routines, as set in [tmc2209 stepper_XX] above."},"gcode_macro _do_error_if_probe_deployed":{"gcode":"\n{% if not printer.probe.last_query %}\n{action_raise_error(\"ERROR: probe still deployed - remove and return to dock.\")}\n{% else %}\nM117 Probe stowed.\n{% endif %}"},"gcode_macro _HEAT_WAIT":{"gcode":"\n{% set MINUTES = params.MINUTES|default(12)|int %}\n{% for i in range(0, MINUTES) %}\nM117 Warming... {MINUTES-i} minute remaining.\n{% for s in range(0, 60) %}\nG4 P500\nG4 P500\n{% endfor %}\n{% endfor %}"},"pause_resume":{},"input_shaper":{"shaper_freq_x":"53.6","shaper_freq_y":"40.4","shaper_type_x":"mzv","shaper_type_y":"mzv"},"resonance_tester":{"probe_points":"\n175, 175, 20","accel_chip":"adxl345"},"probe":{"samples_tolerance_retries":"3","samples":"3","pin":"^PA3","samples_result":"median","sample_retract_dist":"1.5","samples_tolerance":"0.0075","lift_speed":"10.0","y_offset":"25.0","x_offset":"0","z_offset":"10.00","speed":"04.0"},"safe_z_home":{"speed":"20","home_xy_position":"231,350","z_hop":"10"},"gcode_macro M600":{"gcode":"\nM117 Filament Change                                                                 ; Message Text on LCD\n{% set X = params.X|default(50)|float %}\n{% set Y = params.Y|default(0)|float %}\n{% set Z = params.Z|default(10)|float %}\nSAVE_GCODE_STATE NAME=M600_state\nPAUSE\nG91\nG1 E-.8 F2700\nG1 Z{Z}\nG90\nG1 X{X} Y50 F3000\nG91\nG1 E-50 F1000\nRESTORE_GCODE_STATE NAME=M600_state\nM117 Feed Me!"},"stepper_z2":{"rotation_distance":"40","gear_ratio":"80:16","step_pin":"PE2","microsteps":"32","dir_pin":"!PE4","enable_pin":"!PE3"},"gcode_macro UNLOAD_FILAMENT":{"gcode":"\n{% set X_MID = printer.configfile.config[\"stepper_x\"][\"position_max\"]|float / 2.0 %}\n{% if printer.idle_timeout.state == \"Printing\" or printer.pause_resume.is_paused %}\n{% if printer.idle_timeout.state == \"Printing\" and printer.pause_resume.is_paused %}\nSAVE_GCODE_STATE NAME=HUF_state\nM83                       ; relative positioning on extruder\nG0 X{X_MID} Y20 F3500     ; move to area where can easily load filament\nG0 E-50 F1000             ; retract additional filament to move out of melt zone\nG0 E-50 F1000             ; retract additional filament to move out of melt zone\nG0 E-50 F1000             ; retract additional filament to move out of melt zone\nG0 E-50 F1000             ; retract additional filament to move out of melt zone\nG92 E0                    ; reset extruder\nM400\nRESTORE_GCODE_STATE NAME=HUF_state\n{% else %}\n{action_respond_info(\"Pause print first.\")}\n{% endif %}\n{% else %}\n{% if printer.toolhead.homed_axes != \"xyz\" %}\nM117 Performing required homing.\nG28\n{% endif %}\n{% set TEMP = params.TEMP|default(235)|float %}\nM117 Heating nozzle to UNLOAD filamnet.\nG90                       ; absolute positioning\nG0 X{X_MID} Y20 F3500     ; move to area where can easily load filament\nM109 S{TEMP}              ; set hotend temperature and wait\nM83                       ; relative positioning on extruder\nG0 E-50 F1000             ; retract additional filament to move out of melt zone\nG0 E-50 F1000             ; retract additional filament to move out of melt zone\nG0 E-50 F1000             ; retract additional filament to move out of melt zone\nG0 E-50 F1000             ; retract additional filament to move out of melt zone\nG92 E0                    ; reset extruder\nM400\n_PARK_TOOLHEAD\nM18 X Y E\nTURN_OFF_HEATERS\nM117 Finished filament unload.\n{% endif %}"},"mcu rpi":{"serial":"/tmp/klipper_host_mcu"},"gcode_macro QUAD_GANTRY_LEVEL":{"rename_existing":"QUAD_GANTRY_LEVEL_ORIGINAL","gcode":"\nQUERY_PROBE\nM401\nQUAD_GANTRY_LEVEL_ORIGINAL\nM402"},"stepper_z3":{"rotation_distance":"40","gear_ratio":"80:16","step_pin":"PD12","microsteps":"32","dir_pin":"PC4","enable_pin":"!PE8"},"gcode_macro _PARK_TOOLHEAD":{"gcode":"\n{% set Z_MAX = printer.configfile.config[\"stepper_z\"][\"position_max\"]|float -1 %}\n{% set X_MAX = printer.configfile.config[\"stepper_x\"][\"position_max\"]|float -1 %}\n{% set Z_POS = printer.toolhead.position.z + 10 %}\nM118 Parking toolhead.\nG90\nG0 X{X_MAX} Y20 Z{ [(Z_MAX*0.75), [Z_POS, Z_MAX]|min]|max } F3600"},"gcode_macro LOAD_FILAMENT":{"gcode":"\n{% set X_MID = printer.configfile.config[\"stepper_x\"][\"position_max\"]|float / 2.0 %}\n{% if printer.idle_timeout.state == \"Printing\" or printer.pause_resume.is_paused %}\n{% if printer.idle_timeout.state == \"Printing\" and printer.pause_resume.is_paused %}\nSAVE_GCODE_STATE NAME=HLF_state\nM83                       ; relative positioning on extruder\nG0 X{X_MID} Y20 F3500     ; move to area where can easily load filament\nG0 E50 F400               ; prime extruder\nG0 E50 F400               ; prime extruder\nG0 E50 F200               ; prime extruder\nG1 E-0.5 F1800            ; .5mm retraction\nG92 E0                    ; reset extruder\nM400\nRESTORE_GCODE_STATE NAME=HLF_state\n{% else %}\n{action_respond_info(\"Pause print first.\")}\n{% endif %}\n{% else %}\n{% if printer.toolhead.homed_axes != \"xyz\" %}\nM117 Performing required homing.\nG28\n{% endif %}\n{% set TEMP = params.TEMP|default(235)|float %}\nM117 Heating nozzle to LOAD filamnet.\nG90                       ; absolute positioning\nG0 X{X_MID} Y20 F3500     ; move to area where can easily load filament\nM109 S{TEMP}              ; set hotend temperature and wait\nM83 [... cut ...] ":[175.0,348.0,14.3,3188.6322700000283],"speed":27000.0,"extrude_factor":1.0},"extruder":{"pressure_advance":0.05,"target":0.0,"power":0.0,"can_extrude":false,"smooth_time":0.04,"temperature":66.27},"neopixel fysetc_mini12864":{"color_data":[[1.0,0.0,0.0,0.0],[1.0,0.0,0.0,0.0],[1.0,1.0,1.0,0.0]]},"heater_bed":{"temperature":78.23,"power":0.0,"target":0.0},"temperature_fan electronics_fan":{"speed":0.4,"rpm":null,"temperature":32.62,"target":35.0},"heater_fan hotend_fan":{"speed":1.0,"rpm":null}}

Clon1998, 2023-02-28:
Can you provide me the Klipper and Moonraker version of both machines you currently use?
It seems like the value for the neopixel.color_order Config I receive from Moonraker is different now.

For your side it looks like this:
```json
"neopixel fysetc_mini12864": {
"pin": "PC12",
"chain_count": 3,
"initial_red": 1.0,
"initial_green": 0.0,
"initial_blue": 0.0,
"color_order": "RGB"
},
```

While for my printer a neopixel looks like this:
'''json
neopixel fysetc_mini12864": {
"initial_red": 1,
"pin": "EXP1_6",
"chain_count": 3,
"initial_white": 0,
"initial_green": 1,
"initial_blue": 1,
"color_order": [
"RGB"
]
},
'''

ChrisWhite1985, 2023-03-01:
My Mainsail has not been updated in a while.

Voron (not working)
Klipper: v0.10.0-357
Mainsail: v2.1.1

RatOS V-Minion (working)
Klipper: v0.10.0-391
Mainsail: v2.1.2

[20230301-json-voron.zip](https://github.com/Clon1998/mobileraker/files/10857928/20230301-json-voron.zip)

Clon1998, 2023-03-01:
> My Mainsail has not been updated in a while.
> 
> 
> 
> Voron (not working)
> 
> Klipper: v0.10.0-357
> 
> Mainsail: v2.1.1
> 
> 
> 
> RatOS V-Minion (working)
> 
> Klipper: v0.10.0-391
> 
> Mainsail: v2.1.2
> 
> 
> 
> [20230301-json-voron.zip](https://github.com/Clon1998/mobileraker/files/10857928/20230301-json-voron.zip)
> 
> 

Thanks for the file.
Hopefully i can look into this later today.

Can you also provide me the Version of Moonraker, not Mainsail?

ChrisWhite1985, 2023-03-01:
Moonraker

Voron (not working)
v0.7.1-340
RatOS V-Minion (working)
v0.7.1-565

Clon1998, 2023-03-01:
> Moonraker
> 
> Voron (not working) v0.7.1-340 RatOS V-Minion (working) v0.7.1-565

Your moonraker versions (mainly v0.7.1-340) are outdated (I can not even find how old but they must be older than 10+ months) So please update them.
I assume you used an older mobileraker version that did not offer full Neopixel support yet, therefore after your phone auto-updated the app the error started to occur since the app now uses more config information and data to provide better neopixel support!

Clon1998, 2023-03-21:
> Moonraker
> 
> Voron (not working) v0.7.1-340 RatOS V-Minion (working) v0.7.1-565

Any news? Did updating klipper resolve the issue or is there sth. else wrong?

Clon1998, 2023-04-21:
Please reopen if the issue is still present

closed 2023-04-21 by Clon1998