Clon1998/mobileraker #121: Quicker Printer Change

redlinebikeparts, 2023-03-04:
Firstly the app is absolutely amazing. I have recommended it to so many people now. So thank you. 

One thing I was wondering is there a way to make changing printer easier? It would be great if I could just tap on the printer name to change over. I have multiple printers so would be nice to have the option. :) 

[image]

Clon1998, 2023-03-04:
Hey,
glad you like it.

It's a bit of a hidden feature, but you can swipe along the title on most of the pages to fastly switch between printers.
(The title you draw an arrow to)

redlinebikeparts, 2023-03-04:
Ohhh damn that's awesome thank you for letting me know.

closed 2023-03-04 by Clon1998