Clon1998/mobileraker #122: RTSP camera urls not allowed

dlettiere, 2023-03-05:
When inputting a camera url it seems that only http addresses are allowed. 

Is there potential for rtsp support in mobileraker? I have a mix of cameras on my printers and love.mobileraker, but find myself using a browser frontend or native camera app when I want to see the non USB cams.

Jemsbsg, 2023-03-06:
+1 I love your app and the only thing missing for me is rtsp link for webcam !

Clon1998, 2023-03-06:
> 

Well, there isnt any hack to get it working with my implementation right now as I currently only support MJPEG streams.
However, I can look into integrating RTSP.

Jemsbsg, 2023-03-08:
thank you very much !

Xelinor, 2024-02-01:
For what it's worth, I use software called Go2RTC to convert my RTSP stream to WebRTC, which Mobileraker, fluidd, and Mainsail does now support 😁

Anonoei, 2025-02-26:
@Xelinor 
Old issue revive. I ran into the same issue. I configured my cameras in moonraker.conf and mobileraker started using the cameras correctly. 
Here is my config (I also use go2rtc)

[webcam Bed]
location: Bed
icon: mdiRadiatorDisabled
enabled: True
service: webrtc-go2rtc
target_fps: 30
target_fps_idle: 5
stream_url: http://yourprinter.local:1984/stream.html?src=bed
snapshot_url: http://yourprinter.local:1984/api/frame.jpeg?src=bed